OrcaSlicer/OrcaSlicer #9303: MQTT command verification failed. Please update Studio or Handy.

rebsam, 2025-04-11:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

Windows

### OS Version

windows 10

### Additional system information

When using BambuStudio or BambuApp this error doesn't occur

### Printer

BambuLab X1C with AMS

### How to reproduce

Start Orca Slicer
click on machine 
HMS has a red dot
click on HMS the message can be read

### Actual results

it is not possible to send any signal to the printer

### Expected results

no red dot at HMS and everything is working fine

### Project file & Debug log uploads

[debug_Fri_Apr_11_20_13_39_32572.log.0.zip](https://github.com/user-attachments/files/19712582/debug_Fri_Apr_11_20_13_39_32572.log.0.zip)

[special half insert v1.zip](https://github.com/user-attachments/files/19712649/special.half.insert.v1.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

repentsinner, 2025-04-22:
Thanks @MxBrnr. Downgrading FW via Bambu Handy from 01.08.05.00 to 01.08.02.00 seems to have resolved this (for now).

xxxvodnikxxx, 2025-06-05:
Interesting, it seems 0,1.08.02 is already having this "feature" implemented for P1S :/

boysie123, 2025-06-08:
or just roll back to the previous firmware using the handy app, there is noting else in the firmware update according to the release notes so you lose nothing and gain everything. thx bbl for this pointless update. the p1s is not really needing any patching everything it does it does well enough now, unless they mandate it to work with the cloud updating it beyond the previous firmware is now pointless, at least they didn't include it before ppl wanted to buy the new ams, wonder why they did that :)

xxxvodnikxxx, 2025-06-08:
FYI at P1S, 01.08.00 seems to work :))

vadixidav, 2025-06-18:
Does anyone know if it is still safe to use Orca Slicer with the latest firmware in developer mode + LAN mode? It seems to work, but I don't know if they purposefully broke the MQTT protocol in some way that might cause issues at some point. I am using a P1S. For those unaware you can still use the latest firmware if you enable developer mode. It says a lot of scary things about how it allows anything to connect to the device, but it does seem to work. I would like if any developers could comment on whether MQTT incompatibility is a concern in any way.

darebgd, 2025-07-01:
Dear god. I mean TLS/SSL is a must for anything that goes public. This is meant to patch a vulnerability issue that may be exploited to do all kinds of stuff. Put yourself in their shoes: someone is able to get inside your network and crank all your heaters while you're not home. Who'd you blame for that?

All you clueless kids, think about it for a second. They have printers in many homes across the world. Security is a though problem in the world of SW development. However, there might be alternative solutions, like digitally signing client certs and so on... Just imagine their nightmare.

darebgd, 2025-07-02:
I think I made a fool of myself. They did a crappy job if the purpose was to introduce security.
That being said, they could have been more transparent about the dev functionality, like @vadixidav says - are we even getting a proper functionality? 

They could just put a big red button and say "security off, use it at your own risk"
Sheesh... I can't follow up with human stupidity anymore. I thought I am open-minded and rational.

I still think that security is the answer - I'd never dare leaving my printer on during the night with absolutely no protection. That doesn't mean that they needed to break the functionality for the sake of 'security'

vadixidav, 2025-07-02:
@darebgd I understand your previous concerns, and your new comment. However, I think we should focus on the technical issues here. Notably, what is and isn't working and what is the compatibility? I really need to keep using Orca slicer no matter the cost, so I can take care of the security aspect with VLANs etc. For me my main concern is getting some clarity between Bambu Lab and any Orca Slicer developers that the MQTT protocol is documented and being used properly. I am not in the weeds here, but I would hope Bambu Labs would document their protocol and provide us with updated documentation when it changes. If not I think we just need to understand the current state of the open source community reverse engineering. So far everything in Orca slicer works with latest firmware, but I don't know how much longer that will last and it would be good to know when that compatibility breaks.

darebgd, 2025-07-02:
To be honest, I don't think they created two MQTT brokers just to support dev mode. I think that dev mode is just switching off the secure connection, allowing tools without client certificates to connect. I have done no research on this topic, and this is just an educated guess.

The 'scariest' thing that might happen IMHO, would be if they store your 'consent' somewhere so that they can pull it out in case of an RMA attempt.

TicoM1, 2025-07-05:
This sucks. So OrcaSlicer devs don't plan on reaching out to Bambu and represent the Community, but what is their plan then? Their fine with losing all Bambu users in the long run?

asm0dey, 2025-07-18:
@KingGeneral I tried, but in my case it can't import 3mf, saying "File format error"

pddar, 2025-09-02:
enable Developer Mode :https://wiki.bambulab.com/en/knowledge-sharing/enable-developer-mode

kestauta-s, 2025-10-21:
The solution that worked for me was simply disabling the "Use legacy network plugin" option.
After that, everything worked as expected—no other workarounds were needed.

[image]

kestauta-s, 2025-10-26:
Actually no need of developer mode. But you need to downgrade printer formware. The only way to make it work with full functions

MxBrnr, 2026-02-07:
@ianalexis I think this issue can be closed since we know it is not a problem with OrcaSlicer, but a result of Bambu's firmware/cloud server lockdown, and we know the workaround solutions are either a firmware downgrade, or using LAN-Only and Developer mode within the printer's firmware.