OrcaSlicer/OrcaSlicer #9307: Flow Ratio Tests Give Conflicting Results

MxBrnr, 2025-04-11:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

v2.3.1-dev, Build 1b8f798

### Operating System (OS)

Linux

### OS Version

N/A

### Additional system information

Issue is not computer or operating system specific.

### Printer

Issue is not printer specific.

### How to reproduce

Load, slice, and print flow ratio Pass 1.
Do the same for Pass 2 without changing _any_ settings.
Do the same for YOLO, also with no setting changes.

### Actual results

Referencing the photo:
Pass 1 (far left) indicates the flow ratio should between 1.0 and 0.95.
Pass 2 (middle) indicates that 0.95 and anything greater is over-extruded.
YOLO (far right) also indicates the same as Pass 2
All three tests showed a volumetric flow rate of 8.30 for their respective chits shown in the photo, and no settings were manually changed.

Previous calibration of this filament in OrcaSlicer v2.3.0-stable (with default monotonic line as top solid infill pattern) showed a flow rate of 0.98 was ideal, suggesting that the new Pass 1 is correct, while Pass 2 and YOLO are wrong.

[image]

### Expected results

All 3 tests should be consistent with each other in the given results and what flow ratio is ideal.
I would also expect their results to be generally consistent with those of tests from previous slicer versions.

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

ShaneDelmore, 2025-04-12:
That top layer concentric pattern looks strange, is that due to concentric pattern being broken in recent versions of Orca?

henrivdr, 2025-04-12:
> That top layer concentric pattern looks strange, is that due to concentric pattern being broken in recent versions of Orca?

That is intentional in the new version. The last segment printed between the interrupted ones and the contiguous one would bulge out if it does not fit, due to over extrusion.

MxBrnr, 2025-04-12:
> That top layer concentric pattern looks strange, is that due to concentric pattern being broken in recent versions of Orca?

It is actually not concentric, it is the pattern known as `Archimedean Chords`, and is intentional. It is also programmed to print the lines of the pattern in a certain order to facilitate the "ease of use" for the person trying to decipher the results (one of the non-connected sides is printed after the center circle).

ShantanuNair, 2025-04-14:
I have a hunch what's causing it, something I'm working on already, please see below, and let me know if you think it applies here:

In summary, the problem arises because the slicer currently factors in print_flow_ratio in the calculation of _mm³_per_mm for speed capping, but it does not account for filament_flow_ratio—which is applied later in the extruder conversion. As a result, when users follow the documentation (and set filament_flow_ratio based on calibration tests), the effective nozzle flow (i.e. geometry × print_flow_ratio × filament_flow_ratio) is not accurately reflected in the speed capping. This is because the original slic3r design was such that the filament_flow_Ratio was meant to be used only for extruder skipping/over/underfeeding - and at values >1, for all features where the extruder isn't skipping and hence pushing more plastic relative to the model's geometric volume, it will cross the max volumetric speed and overextrude.

Here, you're comparing yolo prints which combine both ratios, to the other calibration test, where just print flow ratio is changed. They don't have similar flow profiles. My [issue](https://github.com/SoftFever/OrcaSlicer/issues/9261) is perhaps the same as yours, and [my PR could maybe be fix](https://github.com/SoftFever/OrcaSlicer/pull/9218) your issue.

ShantanuNair, 2025-04-14:
Reread your issue a couple times. 
- pass 1 vs pass 2: that's odd, they should be the same extrusions. 
- pass 1/2 vs yolo: As explained prior, doing a yolo test with filament ratio 0.95 would result in different extrusions than if directly setting print_flow_ratios. Can you test by setting each yolo calibration pad's print_flow_ratio (under per object settings) to your desired value directly (and set filament flow ratio to 1). What do you see then? How's it compare to pass 1/2. 

Also a screenshot of the flow previews for each slice might be helpful

MxBrnr, 2025-04-14:
Updated to the latest nightly build, b759708 ( as of 7hrs ago).
There is no change to the volumetric flow displayed in the sliced preview, regardless of which flow ratio was changed; the object specific one, or the filament profile. All tests show 8.30 for the -5 chit. Speed capping should not be a factor as the max volumetric flow for this filament is 16 for my setup.

Flow Ratio Pass 1 - filament flow ratio 1.0, object flow ratio 0.95
[image]

Flow Ratio Pass 1 - filament flow ratio 0.95, object flow ratio 1.0
[image]

Flow Ratio Pass 2 - filament flow ratio 1.0, object flow ratio 0.95
[image]

Flow Ratio Pass 2 - filament flow ratio 0.95, object flow ratio 1.0
[image]

YOLO Flow - filament flow ratio 1.0, object flow ratio 0.95
[image]

YOLO Flow Ratio - filament flow ratio 0.95, object flow ratio 1.0
[image]

Actual test print results to follow...

MxBrnr, 2025-04-15:
It appears that some update within the past 10 days has fixed the issue. All three tests (Pass 1, Pass 2, and YOLO) are now reporting consistently with each other, giving the same value for calibration; in this case, 0.92. Looking through the PRs from that timeframe, I suspect it may have been the top surface pattern print order.

Pass 1 (filament flow 1.0)
[image]

Pass 2 (filament flow 1.0)
[image]

YOLO (filament flow 1.0)
[image]

Side-by-side (all at filament flow 1.0, object flow ratio 0.95)
[image]

As it stands now, I will consider this issue resolved, and will close it by the end of the week unless further info arises, or is requested.

closed 2025-04-25 by MxBrnr