OrcaSlicer/OrcaSlicer #9318: Big outer brim also adds to the inside of nearby objects

FelixBuehler, 2025-04-12:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Bambu Lab P1S

### How to reproduce

1. Have some objects close to each other which an inner hole / potential inner brim area
2. Set a brim big enough, to have the brim of one object reach the inner hole of another object
3. Slice; the other object will get an inner brim

### Actual results

Inner Brim appears.

[image]

### Expected results

No inner brim or no brim added to the inside of other objects.

### Project file & Debug log uploads

[Brim bug.zip](https://github.com/user-attachments/files/19718983/Brim.bug.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

Noisyfox, 2025-04-13:
Yes this is a know issue and unfortunately can't be fixed easily without performance impact.

robustini, 2025-06-23:
Same issue here.
I hope a way can be found to solve it, Prusa Slicer does not.

closed 2025-09-30 by github-actions[bot]