OrcaSlicer/OrcaSlicer #9322: ER: Flow rate per object

arhi, 2025-04-12:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

Print in place assemblies can benefit a lot from some of the objects being printed with lower flow rate and some of the objects being printed with higher flow rate.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Existing https://github.com/SoftFever/OrcaSlicer/issues/7684 and https://github.com/SoftFever/OrcaSlicer/issues/8544 ER's would add per feature which is great but all that "per object" can be much more important in many cases for print in place assemblies.

At least global flow rate "per object" would be very useful. Of course per object and per feature would be even better but at least per object flow rate would really make sense... either % from what's in the filament profile or absolute value replacing the filament settings completely

### Describe alternatives you've considered

1. Creating different fillament profiles and asigning different filament for different objects
2. post processing script to add flow rate after object naming script

MxBrnr, 2025-04-12:
This can already be done using the `Split to objects` tool, selecting the individual object, and then setting the "per object global flow ratio" (which is a multiplier of the flow ratio listed in the filament profile).

[image]

arhi, 2025-04-12:
I apologize I'm @$#^_%&*#^ blind :( ... THANKS! .. was looking for it for hours hoping it was there and missed it :(

closed 2025-04-12 by arhi

ShantanuNair, 2025-04-14:
@MxBrnr @arhi Those two flow ratios are not equivalent and don't function the same. The filament flow ratio is typically only a v small deviation from 1, as it's meant to ensure the right amount of plastic gets to the nozzle (fixing under/over feeding to get expected amount of plastic through the nozzle). 

The per object setting is meant to intentionally over/underextrude vs the geomoetric volume. Slice using both flow ratios, one at a time (set one to your ratio value, and the other to 1), and compare them to each other. I think you'd see an unexpected difference in flow rates throughout the print. So in effect, your issue is still quite valid :) We need a print_flow_ratio which is globally applied.