OrcaSlicer/OrcaSlicer #9324: cylinder gemoetry altered by sloping top surfaces

alterego5196, 2025-04-12:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

Windows

### OS Version

Win10

### Printer

Generic Custom klipper

### How to reproduce

Slicing cylinders with sloping top surfaces yields in altered geometry. 
An overhanging inner surface on the ID beginning at the inside edge of the sloping surface, or an enlarged inner diameter, below the mitered end / slant cut to the cylinder, depening on your frame of reference. 
The error is in the surface generation and is consistent regardless of orientation (slant cut to cylinder on plate or 90 face to cylinder on plate).

screen snip of model as generated:
[image]
notice inner walls are 90 to bottom surface. no change in vertical surface. 

screensnip of object as sliced. 
notice inner walls are 90 to build plate. inner diameter changes at mitered cut /sloping face of top of part. 
overhang is consistently 3-4 perimeters from wall of inner diameter. 
[image]

objects and 3mf included below

gcode
[dean_ring.gcode.zip](https://github.com/user-attachments/files/19721922/dean_ring.gcode.zip)

### Actual results

inner diameter of cylinder is enlarged below miter cut/ sloping surface. 

### Expected results

unaltered geomtery after slicing. 

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/19721955/log.zip)
logs

project file and objects
[dean.zip](https://github.com/user-attachments/files/19721960/dean.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

Noisyfox, 2025-04-13:
[image]

These two parameters caused this issue.

Noisyfox, 2025-04-13:
[image]
Use value like this could compensate this issue, however it will make the object bigger while holes smaller.

alidorauxier, 2025-04-13:
This almost seems like another candidate for painting selection. It would be nice to paint the holes you want x-y expanded or 'unpaint' other hole-like areas if desired.

alterego5196, 2025-04-15:
thanks for al your input. i will revisit those settings and check the results.