OrcaSlicer/OrcaSlicer #9327: Adaptive Layer Height Min/Max

brocc28, 2025-04-13:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

Adaptive Layer Height only allows the slicer to determine the minimum and maximum layer height based on the machine configuration, instead of allowing the user to define a specific min/max layer height within the machine configuration limit.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Using Adaptive Layer height only lets you select the quality and a smoothing effect scale. Maybe you want to use adaptive layer heights but you don't want them to be too short or tall. 

Adding the ability to choose a range instead of using the machine configured min and max layer height.

For example, My machine (Creality K1C) can print at a layer height of 0.08mm to 0.30mm, but i don't want want the slicer to apply that broad layer range on my model when using adaptive layer height. For "x" model, I'd like to choose the layer range of 0.14mm to 0.20mm.

dewi-ny-je, 2025-04-13:
I also would like to go thinner upon need, without anything going thicker than what I set in the slicer settings (not machine settings).

Probably a new option should go in slicing settings, to define minimum thickness, the maximum being the one otherwise used without adaptive layers.

kylek29, 2025-04-15:
Would love this. You can somewhat dial it in now by adjusting the sliders just right, but a more user friendly input of just min/max clamping what it can generate would be very helpful.

zexcster, 2025-06-07:
You can modify your printer profile and set minimum and maximum height as you needed (extruder1, etc tab). just save that profile with another name and use it for adapting layers.