OrcaSlicer/OrcaSlicer #9328: Purge Tower Incorrect Purge amounts on first layer

Xedek, 2025-04-13:
I am having an issue, where the color changes on the first layer are purging much less filament into the purge tower than it should, it is ignoring the volumes that i set.  Every other layer is fine.

**Layer 0**
[image]

**Layer 1**
[image]

You can see a still opened issue related to this with no one looking at it here:
> Same problem confirmed here. This is a major bug. What can we do to bring it to attention? 

 _Originally posted by @mb-software in [#8798](https://github.com/SoftFever/OrcaSlicer/issues/8798#issuecomment-2778719264)_

Noisyfox, 2025-04-13:
Try #9235

Xedek, 2025-04-15:
@Noisyfox This did fix my issue.  I compiled the code and then pulled in your changes. Thank you!  I look forward to this being implemented with the next release!

closed 2025-04-15 by Xedek

Noisyfox, 2025-04-15:
FYI you could also download the PR build from the "Checks" tab on the PR page.

PR #9235 (2025-04-20): Fix prime tower wipe volume calculation
Fix #8798
Fix #9080
Fix #9328