OrcaSlicer/OrcaSlicer #9330: Tree support starting in thin air

kisslorand, 2025-04-13:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

3.2.1 - dev

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

all

### How to reproduce

Slicing some objects with curved bottom that need support results in tree supports starting in thin air.

### Actual results

Layer 5:

[image]

Layer6:

[image]

Layer 7:

[image]

Layer 8:

[image]

Layer 9:

[image]

.. and it so continues to add tree support on thin air on consecutive layers

### Expected results

I expect supports to have solid base.

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/19724761/log.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

pgifford, 2025-04-15:
It occurs in 2.3.1-dev (Build 035b047) when using organic and hybrid, and Bambu Studio 2.0.1.50 when using organic (hybrid is fine).

[image]

kisslorand, 2026-10-30:
It is still an issue with nightly (2.5.0 - dev)