OrcaSlicer/OrcaSlicer #9342: Export and Import Preset Bundle is not working properly

viper0078, 2025-04-14:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3

### Operating System (OS)

Windows

### OS Version

Win11

### Printer

SnapmakerA250

### How to reproduce

When I export from "Export preset bundle" by selecting "Printer Config Bundle" and import it on another computer, some parameters are not registered correctly.
What made me notice the issue was the Line Width parameter in the Process section — a value of 0.82 was set.
There may be other parameters that were not exported or imported correctly as well.

### Actual results

That the correct settings are properly applied.

### Expected results

That the correct settings are properly applied.

### Project file & Debug log uploads

[Snapmaker A250 (0.6).zip](https://github.com/user-attachments/files/19730098/Snapmaker.A250.0.6.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

brandngo, 2025-04-23:
I have the same issue. A workaround is to zip the entire orcaslicer config folder from source machine and unzip in the target machine's config folder.

Nexinity7, 2025-05-28:
@brandngo where is this config folder? i had Orca corrupt its config file on a power loss and the import feature is completely broken and not importing any of my configs

closed 2025-09-05 by github-actions[bot]