OrcaSlicer/OrcaSlicer #9351: Regression in Top Surface + Wall Loops Behavior From 2.1 to 2.3

Friendly3D, 2025-04-14:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

macOS

### OS Version

macOS 15.1

### Printer

Bambu X1C

### How to reproduce

In 2.1 and earlier versions (haven't checked 2.2) you were able to use a thin modifier layer on the surface of an object and set wall loops to a large value like 500 to prevent the surface pattern from tracing objects that are on top of it. This helped with unnecessary nozzle movement and leads to smoother overall surfaces, less build up due to movement, etc.

As of 2.3, this no longer works the same way -- it helps slightly, but no matter how high I set the wall loops value, it doesn't have the same effect.

1. Open attached project
2. Slice
3. Go to layer 35
4. Observe

### Actual results

[image]

### Expected results

[image]

### Project file & Debug log uploads

[surface-repro.3mf.zip](https://github.com/user-attachments/files/19740163/surface-repro.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

discip, 2025-04-14:
@Friendly3D
Try this:
Place the Square and Circle exactly on top of the 'Footprint' not even a hair lower.
## distance between Square/Circle and Footprint = 0 mm
Also enable **`Interface shells`** under **`Multimaterial`**.

[surface-repro.3mf.txt](https://github.com/user-attachments/files/19742538/surface-repro.3mf.txt)

henrivdr, 2025-04-16:
I think Interface Shells negates the modifier workaround completely.  But then, I was not aware of the workaround until now 🤔