OrcaSlicer/OrcaSlicer #9359: having issues connecting to the Creality Hi Combo CFS

Slayerhunter25, 2025-04-15:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3

### Operating System (OS)

Windows

### OS Version

windows 11

### Additional system information

Processor	Intel(R) Core(TM) i7-10510U CPU @ 1.80GHz, 2304 Mhz, 4 Core(s), 8 Logical Processor(s)

### Printer

Creailty Hi Combo

### How to reproduce

1. add stl or 3mf file
2. Slice 
3. print via wifi

### Actual results

with my hot name set. 
i can never see my device it give me a 404 error.
i have another ender 3 v3 set up with a co print and have no issues with connecting 

[image]

### Expected results

i send the print to the printer and it gives me an error 

### Project file & Debug log uploads

FO528 printing with extruding error but it never feeds the CFS
or and XD2572 with is print file corrupted 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

[image]

[image]

[image]

robothicc, 2025-04-17:
Same issue here on my Creality Hi without CFS. It gives me a 404 error on the device interface. Also, when I click the "upload and print" button after slicing, the printer starts printing and then instantly terminates the print job. I can only start printing by uploading the file through WiFi and manually starting it through the printer touch screen.

I've contacted Creality Support already, they basically said it's not their problem

Slayerhunter25, 2025-04-17:
Yeah I would have thought the same thing its not creailty as they have
there own slicer which is hard to use..it a orca issue.  They need to
address I hope
I use usb and it prints..I hope they see this and help..

Gerardb92, 2025-05-22:
Hello.

I have the same issue, I can't connect Orca with CrealityPrint because the port used is 9999 (tcp) and Orca not support. Also If you try to connect from browser not connect. I read that Orca use a Creality Print Agent to connect to printer, never directly. 
As long as Creality doesn't offer an API to interact with your printer, it will be impossible.

impeeza, 2025-06-21:
Another information:  Using Creality Print you can access via port 9999 a more complete interface:

[image]

but accessing the printer via 4408 port you get a more restricted interface called «fluidd», which is the same Orca Slicer shows by now:

[image]

impeeza, 2025-06-27:
Hello there,  If I try to send a print to the Creality Hi printer set up on Orca Slicer with Creality protocol and :4808 port, I get the error:

[image]

also the Creality Hi have another interesting URL http://PRINTERIPORNAME/info, that URL answer with a JSON

[image]

I do not know if is useful.

impeeza, 2025-11-09:
any advances?

Slayerhunter25, 2025-11-09:
I still cant connect issue just use a usb all the time

impeeza, 2026-10-30:
Sadly on the latest version 2.5.0-beta build 345efec6

[image]

The problem persist,  will be nice to have access to Creality Hi and X interface and can send directly printings.