OrcaSlicer/OrcaSlicer #9378: Pauses after every layer

dcan999, 2025-04-17:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.1-Dev

### Operating System (OS)

Windows

### OS Version

11

### Printer

Bambu X1C

### How to reproduce

Slice and Print a file, observe printer. 

### Actual results

I tried out 2.3.1-Dev to get the fixed feature of not pre-splitting components when loading a new file. That worked great, however, it seems like at the end, or beginning of a layer, the nozzle just stops in it's tracks. The bed doesn't drop, and it stops extruding, but the nozzle just sits on the same spot on the print, temp remains at setpoint temp, and it melts the area it's sitting on a bit. I didn't notice until the print was almost done but timed it on the last few layers and it seems to stop for 20 seconds before resuming the layer (or starting the next one, I can't read Gcode that well so I can't tell). It did this with a different print too, so I'm not sure if it's a setting I'm missing or it's a bug but I don't see any setting that would make the nozzle just stop in it's tracks with no Z movement for 20 seconds. 

### Expected results

No 20 seconds pauses between layers. 

### Project file & Debug log uploads

[TPUtest_PLA_40m21s.zip](https://github.com/user-attachments/files/19786626/TPUtest_PLA_40m21s.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

dcan999, 2025-04-17:
Update to say, I downgraded to 2.3.0 and it still does it. Not sure what could be happening? I noticed it also does the 20 second pause after the initial prime line before it starts the print. I searched the G code for any G04 dwells or 20 seconds pauses and didn't see any so I have no idea what's going on.

Noisyfox, 2025-04-18:
Did you have timelapse enabled?

dcan999, 2025-04-19:
> Did you have timelapse enabled?

Thanks for the reply, no I didn't, after researching it more, it was just a coincidence that my SD card went corrupt at the same time I installed the nightly build, so I thought it was related but wasn't. I formatted it and it works fine on the latest nightly build.

discip, 2025-04-19:
@dcan999
If your issue is fixed, please don't forget to close it.

closed 2025-04-20 by dcan999