OrcaSlicer/OrcaSlicer #9379: ESP3D comands

kleffa, 2025-04-17:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

[image]

ESP3D is a upgrade and funcionality for some printers. Personally I use a old Ender3 with this.
ESP3D have comands can be used in gcode, like [ESP600] wich means send a notification, also can take pictures, set leds and others functions.
Saddly, OrcaSlicer detetcts brackets on custom gcode and triggers an error, even if ESP3D is implemented as device conection.

### Which printers will be beneficial to this feature?

Marlin

### Describe the solution you'd like

let use "[ESP***]" as a custom gcode, those commands should not be changed. as commands require brackets 
here is a list of commands
https://esp3d.io/ESP3D/Version_3.X/documentation/commands/

basically let [ESP***] be in gcode

### Describe alternatives you've considered

1. moving back to Cura, by now Cura let me set that line on header with no trouble. 
2. make a workaround to replace <ESP600> to [ESP600] a post procesing step

Noisyfox, 2025-04-17:
Use syntax like `{"[ESP600]"}`

kleffa, 2025-04-17:
> Use syntax like `{"[ESP600]"}`

Thanks, it works!!!
not a elegant solution, but it works

closed 2025-05-03 by Noisyfox