OrcaSlicer/OrcaSlicer #9381: Crash after slicing or when previewing sliced model

Appex213, 2025-04-17:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

Windows

### OS Version

Windows 11

### Additional system information

AMD Ryzen 7 9800X3D
64GB RAM
Nvidia RTX 6000 Ada

### Printer

Bambu Lab P1S

### How to reproduce

1. Load project
2. Slice

OR

1. Load sliced GCODE

### Actual results

Slicing (and remaining on preview tab) or importing sliced GCODE both result in crash.

### Expected results

After slicing / importing gcode preview works correctly.

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/19792821/log.zip)
[project.zip](https://github.com/user-attachments/files/19792822/project.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

### Anything else?

Issue seems to be related to previewing complex models after slicing, though at times seems to also occur at seemingly simple models or not occur at complex models as well. Settings that add 'complexity' to the sliced model such as fuzzy skin or arachne often cause it to crash at preview.

Slicing the model and switching back to the prepare tab before the slicing is complete and the preview appears prevent the crash, and switching back to the preview tab will crash immidiately.

I couldn't make much out of the log files myself, but the exception seems to always be 'c0000005 ACCESS_VIOLATION' though the location the fault is attributed to is not consistent; so far I've seen orcaslicer.dll, ntdll.dll and nvoglv64.dll. All are included in logs.zip.

Appex213, 2025-04-19:
> It's an known issue of the latest nvidia workstation drivers. It's openGL driver is somehow broken. Downgrade to version 55x.x and it should work.

Downgraded to driver version 553.62 and previewing works correctly now.

closed 2025-04-19 by Appex213

Noisyfox, 2025-04-27:
Btw, does this works for you? https://forum.bambulab.com/t/bambu-studio-crashes-after-slicing-solved-nvidia-problem/162392

Appex213, 2025-04-27:
> Btw, does this works for you? https://forum.bambulab.com/t/bambu-studio-crashes-after-slicing-solved-nvidia-problem/162392

I don't want to reinstall the bad newer driver to test but seeing as it currently works without changing this setting I would say it's not the cause if this issue.

I also set it to 'on' and 'off' explicitly to see if there is a difference and couldn't see any.