OrcaSlicer/OrcaSlicer #9382: upload print doesnt work anymore

HcKaos, 2025-04-17:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3

### Operating System (OS)

Windows

### OS Version

windows 10

### Additional system information

After updating from 2.2 to 2.3 printer will error out after sending a print over wifi. cant get connection to printer although i can control it over device tab. 

### Printer

creality k1

### How to reproduce

2.3 not working to upload a print, just errors out

### Actual results

print upload does not work over wifi

### Expected results

should fucking work to upload a print

### Project file & Debug log uploads

-

### Checklist of files to include

- [ ] Log file
- [ ] Project file

rileytech, 2025-04-18:
Had this same issue after Windows updated this morning. I ended up completely turning off the Windows firewall, and now everything works as expected.

stephen-elves, 2025-04-22:
I had a similar issue to this with my Qidi Plus 4, further investigation showed that the internal /tmp RAM disc folder was full, I think from a previouse very large print, rebooting the firmware did not resolve it only power cycling the printer.

Essentially "Have you tried turning it off and on again?"

closed 2025-07-29 by github-actions[bot]