OrcaSlicer/OrcaSlicer #9384: Custom Filaments Missing After Opening 3MF Project

amura11, 2025-04-17:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0 Build 70931e5

### Operating System (OS)

Windows

### OS Version

10

### Printer

Bambu Lab P1S

### How to reproduce

1. Have an existing custom filament assigned to the AMS
2. Open OrcaSlicer
3. Connect to the device
4. Open a 3MF project (attached project file is is known to cause the issue)
5. Check the list of filaments in Prepare -> Filament

P.S. I'll try to write some better recreation steps, I can't add custom filaments since this happened to try and recreate it properly

### Actual results

Existing custom filaments now show as **Generic PLA** (or something similar) and **User Presets** is missing presets. Going to the device and viewing the AMS the original presets are listed there correctly.

[image]

[image]

[image]

### Expected results

Opening projects shouldn't wipe out the custom filaments

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/19797865/log.zip)

[Low+Profile+Anti+Vibration+Feet+.zip](https://github.com/user-attachments/files/19797866/Low%2BProfile%2BAnti%2BVibration%2BFeet%2B.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

### Anything else?

Restarting OrcaSlice doesn't do anything to help. I have backups of my configs so I can revert but that doesn't help me when I want to print one of these projects. This also only seems to happen with certain project files. I'm a complete noob so I'm not sure what the differences are, maybe something with custom presets in the projects?

These issues seem similar but don't describe the same behaviour I'm seeing. It's possible it's the same underlying issue with different behaviour:

- https://github.com/SoftFever/OrcaSlicer/issues/9112
- https://github.com/SoftFever/OrcaSlicer/issues/8935

Here's the filament JSON that disappeared:

[CanadianFilaments PLA Basic @Bambu Lab P1S 0.4 nozzle.json](https://github.com/user-attachments/files/19797924/CanadianFilaments.PLA.Basic.%40Bambu.Lab.P1S.0.4.nozzle.json)

amura11, 2025-04-17:
I'm truly a noon indeed, the project must have switched my nozzle setting and I didn't notice. Switched it back and everything is back to normal 😳

closed 2025-04-17 by amura11

coryrc, 2025-07-26:
Ugh, this happened to me too, as you say it makes some sense. It'd be nice to have a warning if you're loading a 3mf for a printer you haven't used before?