OrcaSlicer/OrcaSlicer #9385: Edit Custom G-code window non-resizable / too large

CoppoloElectronics, 2025-04-17:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3

### Operating System (OS)

macOS

### OS Version

macOS 11.7.10

### Additional system information

Model: MacBook Pro (Retina, 15-inch, Late 2013)
Processor: 2.3 GHz Quad-Core Intel Core i7
Memory: 16 GB 1600 MHz DDR3
Graphics: NVIDIA GeForce GT 750M 2 GB, Intel Iris Pro 1536 MB

### Printer

Snapmaker 2.0 A350

### How to reproduce

1. Edit preset filament
2. Select Advanced tab
3. Under Filament start G-code or Filament end G-code, click Edit Custom G-code icon.

### Actual results

The Edit Custom G-code window is oversized and cannot be resized. Placeholder descriptions are only partially visible. Okay and Cancel buttons are not visible or clickable at all.

[image]

### Expected results

The window should be contained within the viewport of the screen and/or allow resizing.

### Project file & Debug log uploads

This issue isn't project specific, it happens with any open project. Likewise, this isn't a crash issue so a log file probably isn't necessary. If a log file ends up being necessary, I am happy to add one to the issue.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

henrivdr, 2025-04-17:
I can confirm that this happens on Ubuntu 24.04 as well.

discip, 2025-04-17:
Second that!

discip, 2025-04-17:
Same for Manjaro GNOME.

Noisyfox, 2025-04-18:
https://github.com/SoftFever/OrcaSlicer/commit/7dd7b301e1a5517cc968d40dbe819eeddba90050

CoppoloElectronics, 2025-04-18:
@Noisyfox, that's unfortunate. Perhaps a more appropriately sized window is the better option here?

As it stands, the Edit Custom G-code window isn't very useful for viewports that lack the appropriate height to display the entire window. These viewport sizes cannot see placeholder descriptions nor can they access the confirm/cancel buttons to accept changes.

This feature could be very useful to me as someone new to Orca that's trying to better understand placeholder naming conventions. A fix that allowed me to see the entire window and accept changes would be great.

Noisyfox, 2025-04-19:
Yeah I agree this is a problem for users with small displays. I'll revise this issue soon.

CoppoloElectronics, 2025-04-19:
@Noisyfox, thank you for doing this! I will give this a try.

PR #9405 (2025-05-09): Make Custom GCode Editor resizable
Fix #9385

FYI @Ocraftyone @discip
I also reverts https://github.com/SoftFever/OrcaSlicer/commit/aef74ab0050783bfaf7d67f531dd6877008e843c (https://github.com/SoftFever/OrcaSlicer/pull/3417#issuecomment-1886793312) because I don't think that really matters, given you are not clicking the plus button while changing the selected variable at the same time. And it's not possible to set the button at a fixed position since it's resizeable now. And the resizability brings more benifits than having a perfect fixed plus button.

Now the dialog can be expanded and shri [... cut]
closed 2025-04-19 by CoppoloElectronics