OrcaSlicer/OrcaSlicer #9388: Hardware acceleration on devices tab & Add a toggle to force load web/device page when using 3rd party print host

CodeMasterCody3D, 2025-04-17:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

So when I am printing with my Bambu x1c and octoprint plugin, I can use the 3rd party print host and I can send prints to my bambu lab x1c (via octoprint plugin), but it will not load up the devices page. Neither windows or Linux. I have an arm64 build I made using the GitHub actions. Adding this functionality is beyond my capabilities and would really love to avoid having to use a browser page opened up as well. (Additional notes: when I use chromium or Firefox on my arm64 Linux system and load up klipper and octoprint it runs smooth as butter I can have 5 pages up and everything with videos playing, but when I am using my ender 3 to print things on my arm64 Linux build  of orcaslicer it's laggy and slow, I am running a Proot system on my android phone that has full hardware acceleration, I've installed many things and all hardware acceleration is working including blender, pretty awesome to have blender on my cell phone. I want to ask in this as well if we can add hardware acceleration support for the devices tab web interface so I can run things smoothly.)

Additional request:
add ability to send custom POST command when uploading via 3rd party print host)(Goal is to slim down my scripts so I my scripts don't have to transfer to the SD card when the bambu octoprint plugin can use the octoprint API combined with its pybambu API so that I can use the  POST command that would send it to the octoprint printers SD card) 

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

I'd like the ability to have a toggle (check box) right under the 3rd party print host's check box, so that I can force the device page to use web UI while using the bambu printers.

### Describe alternatives you've considered

1. First alternative would be to just add an extra tab for viewing web UI while using bambu printer, having a toggle (check box) right under 3rd party print host check box(This would keep the devices page bambu uses separate from the 3rd party print host web UI devices page)
2. Ability to create a copy of the bambu printer profile and make a profile version that will load up the devices page and never would use the bambu network plugin.
3. Ability to just remove bambu network plugin and any functionality to the bambu device page(forceing the web UI instead on the device page)

### Additional context

Please add hardware acceleration while in web UI device page. ( Any web browser is way faster and loads klipper/mainsail and octoprint really well because they use hardware acceleration)

Noisyfox, 2025-04-18:
[image]

Noisyfox, 2025-04-18:
Also we didn't diable HW accel. It's up to libwebkitgtk to determine if it's used, unless you run the app with `WEBKIT_DISABLE_COMPOSITING_MODE=1`.

CodeMasterCody3D, 2025-04-25:
> Also we didn't diable HW accel. It's up to libwebkitgtk to determine if it's used, unless you run the app with `WEBKIT_DISABLE_COMPOSITING_MODE=1`.

Weird, it made it crash with that setting.

Idk what the Google Chrome and Firefox is using at the moment but they run flawless. 

If there is anything different maybe in how it loads it specifically for the android side I'll let you know. I'm about to look it up.