OrcaSlicer/OrcaSlicer #9392: P1S with AMS-Pro and AMS cannot print on FW 01.07.50.16 (Beta version)

Disttrack, 2025-04-18:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

Windows

### OS Version

11

### Printer

Bambulab P1S

### How to reproduce

Update FW to version 01.07.50.16(Beta version): 

## New Features
1. Support multi-plate printing.
2. Supports AMS 2 Pro and AMS HT (Need to update to this version before
connecting AMS 2 Pro and AMS HT)

Serial number and version of AMS-Pro not recognized in Device -> update

### Actual results

When send to print something printer stays in "Preheating heatbed", never start print.

### Expected results

Print the file.

### Project file & Debug log uploads

[FlyingProp Set Quality.zip](https://github.com/user-attachments/files/19811997/FlyingProp.Set.Quality.zip)

[log.zip](https://github.com/user-attachments/files/19812016/log.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

### Anything else?

I have to downgrade printer FW to early version 

[image]

mattold7, 2025-04-18:
I have the same issue

adaMATTium, 2025-04-19:
Same issue, too. I could get the prints to work if I switched it to LAN mode.

pennystockss, 2025-04-19:
Yup same here. Had to roll back FW from the phone app. Gave an AMS mapping error for me.

SyFizz, 2025-04-19:
Hello,

Same error for me here.

Janockel, 2025-04-19:
Same for me. Rollback to latest Beta FW also didn't work.

johnny-bit, 2025-04-19:
Please use this opportunity to reach out to bambu support and explain that Orca Slicer works nicely for you, but not on the new beta firmware - hopefuly they (Bambu Labs) will fix the bug that's causing the issue.

Disttrack, 2025-04-19:
> Please use this opportunity to reach out to bambu support and explain that Orca Slicer works nicely for you, but not on the new beta firmware - hopefuly they (Bambu Labs) will fix the bug that's causing the issue.

But I don't know if is an orcaslicer issue or bambulabs fw issue. I believe that is orcaslicer issue because the AMS2-PRO is new and have more features than "normal" AMS.

johnny-bit, 2025-04-19:
> I don't know if is an orcaslicer issue or bambulabs fw

Simple - if a FW change stops normal operation (such as printing) from starting, but it works fine on previous FW, then it's clearly on firmware.

Firmware may offer new features and orca may not support them - that's not a bug of orca, but a feature request may be needed. But if the new firmware stops basic feature (printing IS basic feature) then it's a firmware bug.

Disttrack, 2025-04-19:
> Simple - if a FW change stops normal operation (such as printing) from starting, but it works fine on previous FW, then it's clearly on firmware.
>
> Firmware may offer new features and orca may not [...]

But with bambustudio all works fine

johnny-bit, 2025-04-20:
Does this firmware update bring along the anti-consumer and absolutely useless "authorization control" under the guise of "security"? If so - have you tried enabling lan-only&developer mode?

Disttrack, 2025-04-20:
> Does this firmware update bring along the anti-consumer and absolutely useless "authorization control" under the guise of "security"? If so - have you tried enabling lan-only&developer mode?

According to the changelog (I've put it in the issue) it doesn't say anything about the security update.

Disttrack, 2025-04-20:
[image]

New changelog that indicate you need new versions of bambustudio and handy app

Disttrack, 2025-04-20:
If instead of printing, you send the gcode file to the machine and then print from the SD card (all from orcaslicer) everything works correctly.

igiannakas, 2025-04-22:
This is the consequence of the firmware locking that Bambu has put in place. Sorry, but I dont think Orca can do anything unless Bambu adjusts what is and isn't allowed with their firmware. 

Options till then:
1. Enable developer lan mode
2. Use their Bambu connect app to send files from Orca to your printer 
3. Use the sd card to print.

Disttrack, 2025-04-23:
> This is the consequence of the firmware locking that Bambu has put in place. Sorry, but I dont think Orca can do anything unless Bambu adjusts what is and isn't allowed with their firmware.
>
> Options [...]

I've already tried using the Bambu Connect app, but the same problem persists.

Also, I haven't seen an option to enable dev mode.

Only works send to printer and print from SD card.

johnny-bit, 2025-04-29:
> Word of caution: do **NOT** update your firmware to 01.08.00.00 to test. I just did, and Handy removed all previous firmware builds from the app for me, so I now cannot downgrade at all.

Well, that sucks... Have you tried the procedure in https://wiki.bambulab.com/en/p1/manual/P1-firmware-update-from-SD-card and getting the https://wiki.bambulab.com/en/p1/manual/P1-firmware-update-from-SD-card 01.07.00.00 ?

Also - I think the network problem is thanks to bambu updating network communication anyway (without mentioning "security" features) and requiring new network plugin... And there are issues with it anyway: https://github.com/bambulab/BambuStudio/issues/6726

Noisyfox, 2025-04-29:
> Handy removed all previous firmware builds from the app for me, so I now cannot downgrade at all.

Whaaaaaaaaaaaaaat??????

NTPanaccio, 2025-04-29:
I just resolved my issue by doing the following:

I was able to get the firmware rollback options by doing the following. I have no idea if this is some sort of official process, or just dumb luck for me, but in case anyone else runs into the issue where old firmware is gone:

1. Follow the steps to enable Offline Update: https://wiki.bambulab.com/en/p1/manual/P1-firmware-update-from-SD-card
2. Perform an offline update to 01.07.00.00
3. Reboot the printer
4. Leave the Bambu beta program in Handy
5. Reboot the printer
6. Old firmware builds should now show up in Handy

Noisyfox, 2025-04-29:
Did the roll out 1.8 firmware for P1S? I haven't received that yet.

Disttrack, 2025-04-29:
> Word of caution: do **NOT** update your firmware to 01.08.00.00 to test. I just did, and Handy removed all previous firmware builds from the app for me, so I now cannot downgrade at all. I basically locked myself out of OrcaSlicer with no recourse. The latest Orca nightly build doesn't resolve the issue, either... not that I thought it would.

I have updated my P1S and yes, I can go back to previous versions from the Handy app.

keithwkx, 2025-04-29:
So does the ams mapping issue still persist in 1.08?

NTPanaccio, 2025-04-29:
> So does the ams mapping issue still persist in 1.08?

It did for me, yes. I had that issue from beta 01.07.50.18 through 01.08.00.00, 100% of the time. Was never able to move past the heatbed preheating stage because of it.

ansonl, 2025-04-29:
The unable to start print occurs when the new firmware is set to LAN mode as well?

NTPanaccio, 2025-04-29:
> The unable to start print occurs when the new firmware is set to LAN mode as well?

I never tried LAN mode. That was going to be my next step had I been unable to downgrade the firmware.

arekm, 2025-04-29:
Note that P1S firmware 01.08.00.00 + OrcaSlicer 2.3.0 and things that work from within Orca:

* video stream
* changing filament types in AMS
* turning off/on led
* homing
* moving toolhead

so I think that printing issue is not due to bambu upcoming (pseudo) security restrictions.

Did anyone compare gcode from bambu studio vs orca when generating gcode for printer with that new firmware?

Disttrack, 2025-04-29:
> Note that P1S firmware 01.08.00.00 + OrcaSlicer 2.3.0 and things that work from within Orca:
>
> * video stream
> * changing filament types in AMS
> * turning off/on led
> * homing
> * moving toolhead
>
> so I [...]

I think it's due to multi-plate printing support or some code change so it can handle the new AMS2. I don't think they've implemented the "security restrictions" in this firmware version yet.

From OrcaSlicer, you can send the gcode and then print it without any problem from the microSD card.

Even though it's sent through the Bambu Connect app, it doesn't work if you send the exported 3mf, so I assume there must be something different there. If it were due to the "security restrictions," it should work from this app, right?

Comparing the gcodes within the 3mf files, they both seem too similar for the problem to be that instead of the other compressed data within the 3mf.

ansonl, 2025-04-29:
From my previous AMS and love experiments with the old P1S firmware:

1. Printing from microsd uses AMS mapping slots from the gcode comment metadata.

2. Printing directly from the slicer uses the slicer_info.config metadata within the 3MF for AMS mapping slot info. 

I haven't downloaded the latest Bambu studio that requires new firmware so not sure what has changed.

Noisyfox, 2025-04-30:
> I think it's due to multi-plate printing support or some code change so it can handle the new AMS2. I don't think they've implemented the "security restrictions" in this firmware version yet.
>
> From [...]

I'll investigate this as soon as I receive 1.8 firmware update on my machine.

arekm, 2025-04-30:
Restart printer doesn't work? (it always helped here for printer to notice new fw)

NTPanaccio, 2025-04-30:
When I enrolled in the beta program, it wasn't until like the 5th or 6th reboot that I finally started seeing beta builds listed in Handy.

arekm, 2025-04-30:
It's now available as offline update, too: https://bambulab.com/en/support/firmware-download/p1

Noisyfox, 2025-05-01:
Finally

[image]

[image]

Noisyfox, 2025-05-01:
Please give #9517 a try and see if it works

Disttrack, 2025-05-01:
> Please give [#9517](https://github.com/SoftFever/OrcaSlicer/pull/9517) a try and see if it works

With this PR I can print again without problems with OrcaSlicer.

I've only noticed one minor issue in my case: no matter how many times I set the PA to slot 2 of my second AMS unit, it clears itself after a few seconds. This only happens with that slot, which is a bit odd.

IetIesAai, 2025-05-01:
testing on Fedora 42, using the flatpak the "new" plugin doesn't see my printer for some reason, the Appimage does see it ("cloud mode") and successfully starts a print through the ams2 pro.

jimmyyuen, 2025-05-17:
I have the same issue

OlivierV78, 2025-05-25:
Same issue with 01.08.01.00

I went back to 01.07.00.00. With this firmware it works with orca 2.3.0

Noisyfox, 2025-06-01:
> > > Please give [#9517](https://github.com/SoftFever/OrcaSlicer/pull/9517) a try and see if it works
> > 
> > 
> > With this PR I can print again without problems with OrcaSlicer.
> > I've only noticed one minor issue in my case: no matter how many times I set the PA to slot 2 of my second AMS unit, it clears itself after a few seconds. This only happens with that slot, which is a bit odd.
> 
> How do I download this PR build? Is it the nightly?

I've update the post to include the download instructions, please take a look.

closed 2025-07-06 by SoftFever

lin-ycv, 2025-07-08:
> Same issue with 01.08.01.00
> 
> I went back to 01.07.00.00. With this firmware it works with orca 2.3.0

how are you able to use 0.1.07 with AMS 2? support for it is only added in 0.1.08?

OlivierV78, 2025-07-09:
@lin-ycv I did not try