OrcaSlicer/OrcaSlicer #9394: github video on release channel not working with firerfox 137.0.2 64bit

rmayergfx, 2025-04-18:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

Windows

### OS Version

Windows 10 22H2 all patches applied

### Printer

no printer 

### How to reproduce

https://github.com/SoftFever/OrcaSlicer/releases

[image]

### Actual results

not working

### Expected results

show video

### Project file & Debug log uploads

none

### Checklist of files to include

- [ ] Log file
- [ ] Project file

pgifford, 2025-04-24:
The next video down, backup.mp4, works in Firefox however.  Both videos are fine in Chrome.

rmayergfx, 2025-07-24:
Firefox 140.1.0esr (64-Bit) still the same issue.

closed 2025-10-31 by github-actions[bot]