OrcaSlicer/OrcaSlicer #9395: runtime org.gnome.Platform branch 46 is end-of-life

mfraser, 2025-04-18:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.1

### Operating System (OS)

Linux

### OS Version

Kubuntu 24.04

### Printer

Creality K1 Max

### How to reproduce

1. Carry out a flatpak update
2. ```
Looking for updates…

Info: runtime org.gnome.Platform branch 46 is end-of-life, with reason:
   The GNOME 46 runtime is no longer supported as of April 17, 2025. Please ask your application developer to migrate to a supported platform.
Info: applications using this runtime:
   io.github.softfever.OrcaSlicer
```

### Actual results

OrcaSlicer is using an outdated version of org.gnome.Platform

### Expected results

Should've kept with AppImage

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Noisyfox, 2025-04-18:
https://github.com/prusa3d/PrusaSlicer/issues/14459

eternal-sorrow, 2025-07-18:
Not stale.

Noisyfox, 2025-07-18:
We have updated to 47 in #9527