OrcaSlicer/OrcaSlicer #9397: Slicing model with many triangles, then modifying layer height and retrigger slicing, causes extremely slow slicing.

TheOneOgre, 2025-04-18:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0 and 2.3.1-dev

### Operating System (OS)

Windows

### OS Version

Windows 11

### Additional system information

i7-12700k
32gb RAM
RTX 3060 TI

### Printer

Bambu P1S (seems agnostic)

### How to reproduce

Load up the 3mf file from below. Perform a "Slice Plate". then change layer height from 0.2 to 0.16 (do not slice yet), and then change it back to 0.2 and slice again. (Could be other settings too, only tested a few and layer height/first layer height seem to cause this.

### Actual results

The time for me, goes from 47 seconds total slice, to 5 2 minutes minutes with no completion. You can see CPU utilization is nowhere near what is was during the first slice. For me that dropped from sometimes 90-100% utilization at moments, to never creeping above 20%. Nothing has changed otherwise except making a setting change, and even changing then reverting does not resolve the issue. Only reloading the project seems to resolve it.

### Expected results

It should re-slice in the same amount of time it did before. No amount of other settings changes seems to bring the speed back except reloading. 

### Project file & Debug log uploads

[slicingtestsavebw.zip](https://github.com/user-attachments/files/19814606/slicingtestsavebw.zip)

Nothing interesting in logs. Just standard startup and 

[debug_Fri_Apr_18_13_08_15_32684.log](https://github.com/user-attachments/files/19814649/debug_Fri_Apr_18_13_08_15_32684.log)

### Checklist of files to include

- [x] Log file
- [x] Project file

### Anything else?

This also happens in Bambu Studio. I tried the latest version v2.0.2.57 and it suffers the same issue. Letting the first slice complete or cancelling it, does not affect the outcome. I tried loading the project into PrusaSlicer to test but I'm not sure it's an accurate test as I couldn't get settings apples to apples.

I tried a bunch of different tests. Adding and removing models, changing the layer height, reverting, then changing something like speed (Changing speed right from project load does no have an effect)

I'm also not exactly sure if the amount of triangles is what is causing it, but I tried this with smaller models and couldn't seem to replicate it as easy as very detailed models.

Noisyfox, 2025-04-19:
Nice catch! I could reproduce this. That's indeed a weird issue. Will take a look soon.

Noisyfox, 2025-04-19:
From what I noticed from the code is that the second slicing needs to clean up the results from previous slicing, which makes it significantly slower. I'll spend some more time on this and see if anything can be improved.

Noisyfox, 2025-04-19:
Could you give https://github.com/SoftFever/OrcaSlicer/pull/7177 a try and see if it helps?

TheOneOgre, 2025-04-20:
> Could you give [#7177](https://github.com/SoftFever/OrcaSlicer/pull/7177) a try and see if it helps?

I would, but tbh I'm not exactly sure how to correctly apply a PR to a build to test and fix. I've tried with other projects and barely scraped by lol. Seems also that you updated that PR to confirm it fixes the issue? Is the testing still necessary? Thanks for your help!

Noisyfox, 2025-04-20:
You can directly download the automatic CI build of the PR, by clicking the "Checks" tab on top of the PR page and you'll find an "Artifacts" dropdown which you could download.

TheOneOgre, 2025-04-20:
> You can directly download the automatic CI build of the PR, by clicking the "Checks" tab on top of the PR page and you'll find an "Artifacts" dropdown which you could download.

Ah, TIL. Yes, #7177 resolves the issue. I had this issue crop up again on another project I have with multiple large models, and trying that with the PR also is resolved, so seems to be fixed with that.

PR #7177 (2025-06-18): Upgrade clipper & improve multi-thread performance
Port all clipper updates from PrusaSlicer, with lots of bug fixes from upstream clipperlib.
Update the clipperlib to the latest version 6.4.2.
Fix #7160
Also ported the memory & performance improvements by replacing std malloc with tbb::scallable_allocator, which fixes #9397
closed 2025-04-20 by TheOneOgre