OrcaSlicer/OrcaSlicer #9409: Wall printing order = Middle / Outer / Inner

uber6, 2025-04-20:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

For models with various levels of overhang, choosing one wall order doesn't work. Even if this can be done using modifiers, this become overwhelming task if there are many overhangs with different directions (outside overhang vs inside overhang). While this might sacrifice the surface quality, by increasing the chance of overhang quality this may be more preferred option. 

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Add a new wall printing order option: Middle / Outer / Inner

ie. If the wall loops is set to 3, print the middle loop, outer loop, and finally inner loop.

Essentially print the outer and inner loop the last when printing wall.

stur0067, 2025-06-16:
I would like to second the need for this. I recently ran into an issue with a shallow curving bowl shape. The inside of the bowl has severe overhangs on the innermost perimeters. It would be nice to not have it attempt to print this innermost perimeter first. Selecting outer wall first would help the inside of the bowl but hurt the outside of the bowl.

stur0067, 2025-12-24:
This not a duplicate of a request for adaptive wall orders. Middle/Outer/Inner would work for >=3 perimeters and would guarantee that both outer overhangs and inner underhangs have a perimeter to attach to.

MxBrnr, 2025-12-24:
The "adaptive" wall ordering would do exactly that. What is suggested here is really just an alternate solution to the same issue.