OrcaSlicer/OrcaSlicer #9418: Need to protect the project from accidental deletion

pi-squared-studio, 2025-04-20:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

If I work in an open .3mf project and accidentally replaced the model, then when saving, this model will remain in the project with the old name. This will lead to the inevitable loss of the old project. So I've already lost several projects.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

To prevent this from happening, I suggest blocking the "Save" button and leaving "Save As..." active with a warning that the physical model has been changed.

pi-squared-studio, 2025-04-23:
The fact is that I sometimes use new empty projects in order to cut different parts with the same settings, when after printing the next one, I can adjust the printed parameters. Therefore, it is easier to remove the old part and cut a new one. And sometimes I open a project, and I can make such an operation in it by mistake. And if I accidentally save it, the wrong part will be saved in the project. It's just that an opened project doesn't visually differ from a new empty program window.

pi-squared-studio, 2025-04-25:
I have been a professional PC user for over 30 years. I work with many different programs, and I know what the Ctrl-S keyboard shortcut is. Moreover, in some programs, I have already automated the use of this combination to preserve intermediate work results... This is partly what causes such problems with saving. But, I think that in the OrcaSlicer program, a warning would not be superfluous if a corresponding warning occurs in an open project if an already saved existing model is deleted from the bed. This warning will not affect other users in any way, as they usually do not do such things. But from the accidental loss of the project, this feature will be useful! 
Then, I can open several project windows on my computer at one time, and an empty window for testing settings, and randomly do certain actions in another window. I believe that if open projects stood out, for example, with a colored capture, then there would be fewer such mistakes.

I'm sorry for taking up time, but if you don't consider this issue relevant, then you can just end this issue. Or tell me, then I'll close this question. Many thanks for OrcaSlicer. This one is the best!

tlhintoq, 2025-04-26:
Mate, I'm not on the dev team.  I'm just a fellow user like you.
Not my place to close this or make any decisions about it.  I've just been trying to talk to you about how to clean up your workflow/habits so you don't cause yourself the problems you are making for you.  Just trying to show you some things you clearly don't know about the program or the OS that would help you out.

I've been trying to politely tip toe around saying "You need to learn how to use a PC better".
But when you start this bit of "a dirty document should have a different colored header" and so on - stuff that no other program has - because you can't seem to keep track of your own work, or own steps... Literally asking for changes that no other program has because you can't track what you're doing...  well then I'll stop tip toeing.

There's no *accidental* about saving changes to a job.  If you make changes and close or new it asks you if you want to save the changes - You literally get a dialog for it.  Click `NO`.  Slow down.  Take your time.  Take a beat and look at your screen when that dialog pops up.  Ask yourself "do I really want to save this doc in this state?" - train yourself to hit no by default instead of just hitting enter on every dialog without thinking.
[image]

Mate just because you've been on a PC for 30 years doesn't mean you're good at it.  Every problem you've described sounds like its of your own making.  Bad practices... unaware of OS standards... Confusing the advice given about `cntl+N` for `cntrl+S`... Getting lost in your own process and work flow.

If your goal is to start with a job so you can carry over those settings to other jobs - Then just save the settings as a profile. Stop using existing documents as a means to save your settings.  Use settings as a way to save your settings.

- You like the settings on a ... articulated lizard then great.  Save those as "articulated".
- Or... Open the job with the settings you like so they get loaded.  Then add a new plate for your temporary work.  You don't have to delete the other models and risk loosing the main job.  Just add another plate and your new models.  Worst case is you save it with extra models but you don't loose the original intent or content
- Or... As soon as you open the example document with the settings you like do an immediate `Save As...` with the name "delete me" and put it on the desktop.  Now you can do 100 different things and not risk the doc you opened for its preferred settings.  Make `Open` then `Save As..` your instinctive process every morning.
- Or... Open the job with the settings you like so they get loaded.  Then immediate tell it `NEW` and transfer those settings to the new job.  You get a big dialog offering to carry over those settings from the opened doc to the new one.
[image]

In other words: Stop changing the jobs you are so mad about being changed and saved.  Just don't do it.

pi-squared-studio, 2025-04-26:
Yes, great! But no other program makes it easy to confuse an existing project with a new empty one. And the fact that projects always take a long time to create, and many users, I'm sure, mostly of them follow the simple path - open a new window, place a new model and make a g-code out of it. It's easier and more convenient this way. The program allows you to do this. Difficulties begin when saved projects with a completely different logic of work appear. And you also need to keep in mind that project settings are not always the same as system settings, although they may be called the same way.