OrcaSlicer/OrcaSlicer #9420: Notification if the use order of the walls construction is Inner/Outer/Inner and only 2 walls are used

pi-squared-studio, 2025-04-20:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

As in title...
The Inner/Outer/Inner mode is used to enhance the visual quality of the exterior walls and at the same time give them increased durability. This works when there are 3 or more wall loops and when the Outer wall is not in contact while printing with any of the Inner ones. If we print only 2 walls, then the Inner wall will be printed first, then the Outer, which in some projects will degrade the visual quality. The user may simply not notice such a trick and get a different result from the desired one.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

I think that it is necessary to introduce a notice when printing in only 2 walls with this complex order is used. and refine the algorithm with this choice so that it best matches the desired result.

igiannakas, 2025-04-21:
It defaults to outer inner which matches the intent - printing an external perimeter without a neighbouring one. Also this is the same default when not more than 2 perimeters can fit in an island.

pi-squared-studio, 2025-04-22:
By fact the printing is done by inner-outer order (there inner is most middle, the first in the range, if used 3 walls and i/o/i mode), that may be worst and illogical idea if I want to preserve the shine of models printed with silk PLA.

igiannakas, 2025-04-23:
Swapping from IOI to inner outer will cause a print defect (wall bulge) on that layer. 

I’ve worked really hard to avoid this scenario as it causes visible surface discontinuities. 

[image]

The shine of models printed with silk PLA shouldn’t change when the algorithm falls back to outer inner.

Post images of your print please and also a set of pictures from your filament cooling settings.

pi-squared-studio, 2025-04-26:
Yes, that's right. When the inner wall comes first, then the outer, the shine of the walls becomes noticeably worse.
I did a little test here. I printed 3 Voron cubes 2x2x2 cm in one pass. The first one has 2 walls, the order is inner-outer. The second one has 2 walls, the order is outer-inner. The third one has 3 walls, inner-outer-inner.
[image]
I must say right away that the photo was taken in extreme conditions when the light falls from above. The cubes are turned to the side where the transition from blue to green of the bicolor silk PLA was.
In reality, all prints are glossy, smooth and uniform. The mechanics of the printer work flawlessly, so it shows the defects that occur when choosing wall laying.
[image]
The third variant is the best in execution. There are an order of magnitude fewer visual defects.
The most disgusting is the first one. there is a random uneven laying of the walls. The fact is that the outer wall interacts with the already printed inner wall, and there is no need to talk about clear dimensions. I get the same result if I/O/I mode is selected and I switch from 3 walls to 2, for example, to save plastic or printing small parts where having only 2 walls would be preferable.

igiannakas, 2025-04-26:
Indeed IOI will give you the best surface quality.

However your printer is in desperate need of tuning. Those artefacts shouldn’t be that noticeable.

pi-squared-studio, 2025-04-26:
No, the photo was taken in special conditions to emphasize the difference in the way the walls are laid. 
Here is a test model showing that everything is OK with the mechanics of the printer. 

[image]

Everything is so good that you can see a small permanent ripple from the steppers or gears of the extruder, which is difficult to correct. Here you need to adjust the printing speed, and that's a completely different story. :)

igiannakas, 2025-04-26:
Ah great!

Indeed inner outer inner is my personal go to print mode for the vast majority of prints I do.

When there are prints with overhangs greater than 60 degrees I revert back to inner outer but with precise wall enabled. Reduces the appearance of those inconsistencies a bit but not the same as IoI.