OrcaSlicer/OrcaSlicer #9421: Crash when opening in Ubuntu 24.4.02 LTS

vdubedout, 2025-04-20:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

Linux

### OS Version

Ubuntu 24.4.02 LTS

### Additional system information

CPU
```
Architecture :                              x86_64
  Mode(s) opératoire(s) des processeurs :   32-bit, 64-bit
  Address sizes:                            43 bits physical, 48 bits virtual
  Boutisme :                                Little Endian
Processeur(s) :                             16
  Liste de processeur(s) en ligne :         0-15
Identifiant constructeur :                  AuthenticAMD
  Nom de modèle :                           AMD Ryzen 7 2700 Eight-Core Processor
    Famille de processeur :                 23
    Modèle :                                8
    Thread(s) par cœur :                    2
    Cœur(s) par socket :                    8
    Socket(s) :                             1
    Révision :                              2
    Frequency boost:                        enabled
    CPU(s) scaling MHz:                     51%
    Vitesse maximale du processeur en MHz : 3200,0000
    Vitesse minimale du processeur en MHz : 1550,0000
    BogoMIPS :                              6399,70
    Drapaux :                               fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush mmx fxsr sse sse2 ht syscall nx mmxext fxsr_opt pdpe1gb rdts
                                            cp lm constant_tsc rep_good nopl nonstop_tsc cpuid extd_apicid aperfmperf rapl pni pclmulqdq monitor ssse3 fma cx16 sse4_1 sse4_2 movbe popc
                                            nt aes xsave avx f16c rdrand lahf_lm cmp_legacy svm extapic cr8_legacy abm sse4a misalignsse 3dnowprefetch osvw skinit wdt tce topoext perfc
                                            tr_core perfctr_nb bpext perfctr_llc mwaitx cpb hw_pstate ssbd ibpb vmmcall fsgsbase bmi1 avx2 smep bmi2 rdseed adx smap clflushopt sha_ni x
                                            saveopt xsavec xgetbv1 clzero irperf xsaveerptr arat npt lbrv svm_lock nrip_save tsc_scale vmcb_clean flushbyasid decodeassists pausefilter 
                                            pfthreshold avic v_vmsave_vmload vgif overflow_recov succor smca sev sev_es
Virtualization features:                    
  Virtualisation :                          AMD-V
Caches (sum of all):                        
  L1d:                                      256 KiB (8 instances)
  L1i:                                      512 KiB (8 instances)
  L2:                                       4 MiB (8 instances)
  L3:                                       16 MiB (2 instances)
NUMA:                                       
  Nœud(s) NUMA :                            1
  Nœud NUMA 0 de processeur(s) :            0-15
Vulnerabilities:                            
  Gather data sampling:                     Not affected
  Itlb multihit:                            Not affected
  L1tf:                                     Not affected
  Mds:                                      Not affected
  Meltdown:                                 Not affected
  Mmio stale data:                          Not affected
  Reg file data sampling:                   Not affected
  Retbleed:                                 Mitigation; untrained return thunk; SMT vulnerable
  Spec rstack overflow:                     Mitigation; Safe RET
  Spec store bypass:                        Mitigation; Speculative Store Bypass disabled via prctl
  Spectre v1:                               Mitigation; usercopy/swapgs barriers and __user pointer sanitization
  Spectre v2:                               Mitigation; Retpolines; IBPB conditional; STIBP disabled; RSB filling; PBRSB-eIBRS Not affected; BHI Not affected
  Srbds:                                    Not affected
  Tsx async abort:                          Not affected
[... 311 lines cut ...]
### Expected results

App do not crash

### Project file & Debug log uploads

[debug_Sun_Apr_20_0.zip](https://github.com/user-attachments/files/19825005/debug_Sun_Apr_20_0.zip)

Not much in logs
```
[error]	2025-04-20 06:51:34.371466[Thread 0x00007965c59ff6c0]:Unable to open connection to spacenavd
[error]	2025-04-20 06:51:34.374412[Thread 0x00007965c49fd6c0]:DBus Request name Error: org.freedesktop.DBus.Error.ServiceUnknown
[error]	2025-04-20 06:51:34.374433[Thread 0x00007965c49fd6c0]:Dbus Messages listening terminating.
[error]	2025-04-20 06:51:34.986111[Thread 0x000079663a3d43c0]:calc_exclude_triangles:Unable to create exclude triangles
```

### Checklist of files to include

- [x] Log file
- [x] Project file

### Anything else?

May be a duplicate of https://github.com/SoftFever/OrcaSlicer/issues/9243
But it does not crash at the same time (same OS, same flatpak though)

vdubedout, 2025-04-20:
Definitively a duplicate of https://github.com/SoftFever/OrcaSlicer/issues/7210

To fix it, append :7125 to your IP Address

As it crashes at each start-up you need to update the json file in the cache folder

So for flatpak, go in /home/<user>/.var/app/io.github.softfever.OrcaSlicer/config/OrcaSlicer/user/default/machine
for AppImage, /home/<user>/.config/OrcaSlicer/user/default/machine

[image]

closed 2025-04-20 by vdubedout

soaresluciano, 2025-04-20:
> Definitively a duplicate of [#7210](https://github.com/SoftFever/OrcaSlicer/issues/7210)
> 
> To fix it, append :7125 to your IP Address
> 
> As it crashes at each start-up you need to update the json file in the cache folder
> 
> So for flatpak, go in /home//.var/app/io.github.softfever.OrcaSlicer/config/OrcaSlicer/user/default/machine for AppImage, /home//.config/OrcaSlicer/user/default/machine
> 
> [image]

That works, however adding the IP on "Device UI" field (print_host_webui) causes the same issue.