OrcaSlicer/OrcaSlicer #9424: Inner walls incorrectly treated as overhangs (fan + speed) — OrcaSlicer 2.3

glstas, 2025-04-20:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3

### Operating System (OS)

Windows

### OS Version

Windows 10

### Printer

Bambu Lab A1 mini

### How to reproduce

🐞 Bug Report: Inner walls without overhangs use overhang fan speed and reduced print speed (OrcaSlicer 2.3)
Slicer: OrcaSlicer 2.3
Material: PETG
Model file: (attached)

📌 Issue summary:
In OrcaSlicer 2.3, inner vertical walls with no overhang geometry are incorrectly treated as if they require overhang cooling and speed settings.

This leads to:
Unnecessary 100% fan on straight inner walls
Slower overhang print speeds being applied
Negative effects on print quality (e.g. weak layer bonding for PETG)

🔍 Evidence from OrcaSlicer:
1. Screenshot: Fan speed preview

[image]

Shows inner vertical walls (not overhangs) printed with full fan speed

2. Screenshot: Speed preview

[image]

Confirms reduced overhang print speed applied to these walls

3. Screenshot: Cross-section view

[image]

No geometry justifying overhang treatment

4. Model file (.3mf) — used for slicing

✅ Expected behavior:
Only actual overhangs should be assigned high fan speed and reduced speed
Inner walls should use standard settings

❌ Actual behavior in OrcaSlicer:
Inner walls (non-overhangs) get overhang-specific fan and speed
Leads to unnecessary slowdown and excess cooling

🆚 Comparison with Bambu Studio 2.0.2.57:
The same model sliced in Bambu Studio is processed correctly:
Only outer overhangs get overhang cooling
Inner vertical walls use normal fan speed and print speed

5. Screenshot: Bambu Studio fan speed preview — correct cooling zones

[image]

7. Screenshot: Speed preview — inner walls printed at normal speed

[image]

🧪 Tested on:
OrcaSlicer 2.3
Bambu Studio 2.0.2.57
Material: PETG
Printer: Bambu Lab A1 mini

### Actual results

Inner walls (non-overhangs) get overhang-specific fan and speed
Leads to unnecessary slowdown and excess cooling

### Expected results

Only actual overhangs should be assigned high fan speed and reduced speed
Inner walls should use standard settings

### Project file & Debug log uploads

[Overhang test.zip](https://github.com/user-attachments/files/19825530/Overhang.test.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

igiannakas, 2025-04-22:
They are above a sharp overhang which means their probability of curling upwards is significant. Hence they are printed slower.

If you want this disabled, disable curled overhang detection.

glstas, 2025-04-24:
Just to add additional info:

The overhang angle is the same (30 degrees) on all the walls, but the print speed is different. Because of that, the print quality looks uneven.

On the printed model, you can clearly see that some walls have different curves and surface finish, even though the shape is the same. It looks like the slicer uses different speeds for no clear reason.

[image]

[image]

igiannakas, 2025-04-24:
I need to investigate it when I get some time - maxed out with work at the moment.

It would be helpful if you posted a full screenshot including the overhang settings used for the above.

igiannakas, 2025-04-24:
Also why do you have partially bridged in the counterbore holes? They should only be enabled by exception where the model demands it (and it’s better if fixed in cad).

igiannakas, 2025-04-25:
As stated in my first comment, this is happening due to the curled overhang detection algorithm. The internal perimeters are slowing down to prevent the overhangs from continuing to curl upwards.

You can disable that as below

[image]

However it is not advisable to do so. A better way would be to tweak your settings to match how much the printer actually needs to slow down for printing overhangs and bridges. In your profile you've set the bridge speed to 10 mm/sec which is excessively low.

Also try increasing the overhang speeds a bit - something like 0-80-50-30 would work better.

[image]

Please close issue as this is working as intended.

closed 2025-04-25 by glstas

igiannakas, 2025-04-25:
That issue you mention on the benchy roof is caused by the extra perimeters on overhangs setting. It creates extra perimeters in areas where bridges are, instead of bridges. In some very specific cases it benefits you but in general it's best to be left as default.

glstas, 2025-04-27:
@igiannakas 
This issue with different speeds across layers is still bothering me.
I recently did a test on a new cool plate bed and noticed the same thing again:

[image]

I had done several tests earlier, and it’s true that enabling “Slow down for curled perimeters” does improve the model overall.
However, the different speeds across layers are still clearly visible on the model walls.

I used the standard settings as you suggested, changing only some small details that shouldn’t affect overhangs:

[image]

[image]
[image]
[image]

[Cool Plate Test.zip](https://github.com/user-attachments/files/19930682/Cool.Plate.Test.zip)

Maybe this is still an actual issue and it should be fixed?
Would love to hear your thoughts!

igiannakas, 2025-04-27:
No it’s not a bug - it’s settings.

You’re slowing down too much for overhangs. Increase the min speed to 0-60-40-30 instead of 0-50-30-10

[image]
 The red ones circled above.

You can also disable curled perimeter detection if it’s not needed in that model.

igiannakas, 2025-04-27:
The problem is your printer is printing the regular perimeters too fast and is not melting the filament thoroughly.

Slow down its regular perimeters to around 120-140mm/sec and increase the overhang speeds as I e highlighted  above

glstas, 2025-04-28:
Hi, thanks a lot for your reply!

About this part:
> "Also the bulge in the corner when going slower indicates your PA tuning is off."

I had already tested earlier with **freshly calibrated PA** and also tried using the **automatic PA calibration** that the printer performs before the print — the results were quite similar.

Regarding speeds:  
I also tried tests with **lower perimeter speeds** (100 for outer wall, 150 for inner wall) and **higher speeds for overhangs**, but in my filament settings the **Maximum Layer Time** was set to 4 seconds.  
Because of that, the actual print speeds on the walls were limited to around 50 mm/s or even lower.

---

Right now, I’m doing a **new test** with your recommended speeds and **“Slow down for curled perimeters”** enabled.  
I also disabled the **Maximum Layer Time** setting to avoid speed limits.

With these new parameters:
- Top layers: wall at 120 mm/s, overhang at 30 mm/s, layer time around 2 seconds
- Lower layers: wall at 120 mm/s, overhang at 66 mm/s, layer time around 1.5 seconds

[image]

[image]

[image]

[image]

[image]

---

Honestly, I didn’t like the result — so I decided to **recalibrate PA again**.

[image]

Here I realized that the **default acceleration values** in default 0.2 profile are too high for a proper PA calibration.  
So I reduced acceleration and re-ran the PA test:

[image]

I selected **PA = 0.05** based on the second test.

---

Then, I reprinted the test model with:
- Lower acceleration
- Recalibrated PA
- Your recommended speed settings

[image]

[image]

[image]

[image]

[image]

**But for me, the result actually got worse** — the corner after the overhang was much more rounded than before.

---

Maybe there’s some other issue in my setup?  
Are you sure that these speed settings should normally improve the result?  
I'd be happy to test more if you have ideas!

glstas, 2025-04-28:
[Cool Plate Test_1.zip](https://github.com/user-attachments/files/19943639/Cool.Plate.Test_1.zip)

igiannakas, 2025-04-28:
Here's my test.

[Model.3mf.zip](https://github.com/user-attachments/files/19946169/Model.3mf.zip)

[image]

[image]
[image]

Dont forget you're pretty much torture testing your printer with this!

Im pretty happy with how it looks. 

I'd suggest re-tune your PA with a bit more granularity - 0.001 step instead of 0.01 step and at your external print speed velocity.

You can see my settings in the attached file.

igiannakas, 2025-04-28:
Btw the curling inwards is due to inadequate cooling. This is a small model. You can drop the material temp by 5-10C and ensure your fan is at 100% constantly as you’re getting material overheating and curling upwards.

glstas, 2025-04-28:
I'm just curious how the G-code is generated for this parameter, but I couldn't find it clearly in the code:
https://github.com/search?q=repo%3ASoftFever%2FOrcaSlicer+slowdown_for_curled_perimeters+language%3AC%2B%2B&type=code

It seems like it might be handled somewhere around here, but the commits are quite old:
https://github.com/SoftFever/OrcaSlicer/blame/9913736b41f311af71200577ca7bc0d03d5c91a4/src/libslic3r/PerimeterGenerator.cpp#L708

glstas, 2025-04-28:
Yes, thanks a lot for your reply!
Your model turned out absolutely perfect at those speeds.
I'll keep digging into it and experimenting further!

igiannakas, 2025-04-28:
> I'm just curious how the G-code is generated for this parameter, but I couldn't find it clearly in the code: [...]

It’s in the estimate extrusion properties function - check the PR I had made ages ago to implement this originally here: https://github.com/SoftFever/OrcaSlicer/pull/2056

igiannakas, 2025-04-28:
One feature you may find useful is this: https://github.com/SoftFever/OrcaSlicer/pull/2161

Extended to apply only to external perimeters here: https://github.com/SoftFever/OrcaSlicer/pull/7399

This can help the extruder and PA respond better as it reduces the instantaneous acceleration requests to the extruder.

I’m printing one more with it on to show you the outputs

igiannakas, 2025-04-28:
[ERS test.3mf.zip](https://github.com/user-attachments/files/19947195/ERS.test.3mf.zip)

Using extrusion rate smoothing (ERS):

[image]
[image]

Minor differences in my case but ever so slightly sharper corners. 

ERS on the left - Disabled ERS on the right:
[image]
[image]

50% of the battle is cooling - make sure it is adequate. The other 30% is PA tuning and the last 20% is ERS.
[image]

sglushko, 2025-04-30:
Thanks for the detailed explanations — I’ll have some free time soon and will try to tackle this!
I’ve been looking into the issue and found that it could also be caused by mechanical problems, like loose belts or poorly lubricated rails. My printer is almost 5 months old, and I’ve only lightly lubricated it twice, but haven’t checked the belts or other components. I plan to service the printer following the manufacturer’s manual and at the same time get a better understanding of the mechanics.
I still don’t fully understand how the belt tensioning works on my A1 Mini. Also, I found out that input shaping in Marlin doesn’t perform as well as it does in Klipper, which might also contribute to the ringing or corner artifacts.

glstas, 2025-05-03:
Thanks for the follow-up! Just to clarify — sglushko is also me, I just forgot to switch accounts 😅

Here’s a summary of what I tried so far:

Lubricating the rails and tightening the belts didn’t noticeably improve print quality, but at least I’m confident everything is well-maintained now.

Drying the filament reduced surface roughness a bit, but didn’t help with the corners.

Enabling ERS made things better, though not completely.

I noticed your model includes infill and three perimeters — this significantly improved the quality in my own tests.

I also suspected cooling was the issue. I wasn’t seeing improvements even with 100% fan speed, so I added a minimum layer time of 5 seconds, which gave similar results to using infill and extra walls.

But I’m still seeing corner artifacts, and now I believe I’ve been tackling the problem from the wrong angle. It’s likely that incorrect PA settings are to blame.

glstas, 2025-05-03:
I noticed that filament flow around corners is not symmetrical — one side before the corner looks thinner and tighter, while the side after the corner is slightly bulged. Here's a close-up photo:
[image]
What could be causing this kind of asymmetric flow behavior?
Thanks for any advice!

igiannakas, 2025-05-04:
Too high Pa causes this artefact. Under extrusion before the corner and over extrusion after. The right PA is where no gaps are present and the corner is sharp.

Have a read through this website: https://ellis3dp.com/Print-Tuning-Guide/articles/pressure_linear_advance/pattern_method.html