OrcaSlicer/OrcaSlicer #9429: Serious Y-Axis layer shift after slicing model with Orca

Veillax, 2025-04-20:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

Windows

### OS Version

Windows 11

### Additional system information

CPU: Intel Core i5-10300H
GPU: Nvidia GeForce GTX 1650 Mobile
RAM: 24GB 2997 MT/s SODIMM DDR4 

### Printer

Ender 3 S1 w/ Klipper Firmware

### How to reproduce

Import 2 objects, in this case an 20mm XYZ Calibration Cube and a purge line
Enable Sequential Printing
Slice Model
Print Model

### Actual results

I am unsure if those steps are the cause of the layer shifting, however it ends up printing with a noticeable amount of layer shifting along the Y-Axis.

### Expected results

Objects should print just fine

### Project file & Debug log uploads

[proj.zip](https://github.com/user-attachments/files/19827060/proj.zip)
No log files pertaining to day of use

### Checklist of files to include

- [ ] Log file
- [x] Project file

### Anything else?

Issue only pertains to Orca Slicer, when slicing the same objects in Prusa, they print just fine, Layer shifting is only along the Y-Axis, not noticeable or occurring along other axes

tlhintoq, 2025-04-23:
Layer shift is not caused by the slicer beyond maybe you trying to print faster than your machine can handle - but generally firmware overrides that.

Layer shift is usually the nozzle catching on something like Zhop pimples, or build up at line crossing from using grid infil etc.

Slow your printing down a lot while testing.
Make sure you've leveled well so you aren't colliding with the print in future layers
Watch the print and listen.  Do you hear the nozzle clicking or twanging as it moves around the job and hitting walls and infil?

Veillax, 2025-04-28:
I usually print at 50% speed, I haven't heard or seen the nozzle hitting any walls or infil, and I tried the same model through Prusa Slicer and it works just fine

TowTungTongue, 2025-04-29:
Installed Orca last year but every print layer shifted so violently i thought it would rip my y-belt in two. Have been using Prusa and Creality slicers just fine. Installed Orca again last week and exactly the same thing happened, googling 'orca layer shift' shows it is a real issue.

solidus0079, 2025-06-01:
I've had this happen several times, and it's not the belts.  Well one time it was, but the other times it wasn't.  It seems to happen in some of the gcode files saved after the "fix model" function is used.  It's just a pattern I have noticed.

Mike-Flexicon, 2025-08-15:
For me , the option "Reduce infill retractions" is what I believe causes this to happen for me. It's definitely the nozzle colliding that causes the layer shift for me, it just seems to only happen when that option is clicked on. I assume it's because I'm printing the infill too quickly and there's some shmutz sticking up that it hits.   This is on an original K1 max with several modifications.

solidus0079, 2025-08-16:
That's definitely not what's happening to me, Mike.
If it was a collision or a loose belt the shifting would be more sporadic and only affect a layer, as it'd still keep trying to print where it should be.
When this happens to me, it's like suddenlt there's a new X/Y home position and will just keep printing in the "new" position.  If you imagine a 3D model and cutting it across into two pieces with the top shifted bit that's what it's trying to create.

OriginalCanadian, 2025-09-05:
Had this issue, played around with slicer settings to fix. 
I believe it has to do with the layer change g-code that exists in the default klipper printer presets. 
Specifically "G92 E0" before layer change. Looks like it slowly shifted the Y axis every layer.
Removed this g-code and suddenly my print ceased shifting.

zatchbot, 2025-09-24:
I am having the same issue as reported above. Y axis shifts about 2mm forward at the same layer every time. I can confirm that OrcaSlicer 2.3.1 is not generating gcode correctly on multi object prints as the upper layers of the smaller object. Here is a side by side of an OrcaCube sliced in both Prusa and Orca. You can see the error clearly.

[image]

[image]

I am currently testing these two files back to back to document results.

zatchbot, 2025-09-24:
The error occurs when the Precise Z Height is enabled. Example gcode attached.

[gcodeExamples.zip](https://github.com/user-attachments/files/22524163/gcodeExamples.zip)

malhar-c, 2026-01-21:
This is interesting, I was having the exact same issue on my modified Ender 3 Neo on both axes how I align the model, in this example I aligned it in the X axis, so the X skips steps here, just for context, I have put more than 1000 hours on it after Klipper conversion never had any problem with layer shifts as I cap my accel to 3k and speeds to 300 max, whereas it can easily do 10k accel (even 15k on the X-axis) and 500mm/s speed. 

Few months ago I switched to Orca slicer (2.3.1) and for the first time I had to deal with an issue like this, for the whole time I was thinking that the nozzle is hitting the part, so I kept on raising my Z hop height, but then I figured that the nozzle was never hitting the part in the first place, there was a brief Jerk (Stop-Start travel move) of the X axis that was causing the step skipping, but it is impossible as I have my accel set wayyy low than the max and it's hard capped in slicer, here is a slo mo video of it : https://youtu.be/Nq724JYxZ14

**Potential Fix?:** I tried to replicate this by reducing Jerk values, speed, even copying those exact same GCODE lines and put in my klipper console, with crazy accel values upto 15k but never able to replicate it, what ended up fixing the issue was just enabling _**Arc Fitting**_ inside Orca slicer to generate G2/G3 commands, but as Klipper doesn't support G2/G3 out of the box I also had to enable the gcode_arc section inside my printer cfg file.

I'm attaching the 3mf file here for testing if anyone wants, the problem begins after layer 35, the exact move is 800 on layer 36, and subsequent layers also have a similar movement where the axis travels at 300mm/s stops and then again starts travelling at 300mm/s, this is where the skip happens, it's not consistent tho what makes me think it's a stall and catch-up game happening, although I don't know where. it's got to be the printer mcu as my Pi is a Pi 4-B 8GB, with proper cooling and everything it barely spikes in resource usage, so no way there is a bottleneck from Klipper side:
Before Fix: [Model with issue](https://drive.google.com/file/d/1-uC0g0gYuvSxRR0nt25VasQ6IhrKMTKq/view?usp=sharing)
After Fix: [Arc Fitting Enabled](https://drive.google.com/file/d/1H-uYm37shIIyIn3ONhgamUF-hH00kEpV/view?usp=sharing)