OrcaSlicer/OrcaSlicer #9433: Auto arrange -  margin from the edge

Gnierat, 2025-04-21:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

Problem with temperature at the edges of the table.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

During automatic setting option add margins option from table edges.

discip, 2025-04-22:
@Gnierat
What about manipulating **`Excluded bed area`**? 😊

[image]

Gnierat, 2025-04-22:
Yeah, but sometimes you need that extra space.