OrcaSlicer/OrcaSlicer #9443: Pretty large multi color issue with support for PLA for 0.09 top distance or lower

CallmeBigfat, 2025-04-21:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

Windows

### OS Version

Windows 11

### Additional system information

RTX 4080 super
32GB RAM
Intel 14TH Gen i9

### Printer

bambulab x1c / p1s

### How to reproduce

1. Add something that is 3 multi colored (plus support pla) on an angle
2. Select Sup.PLA as interface layer
3. Set Top Z distance lower than 0.09   (I need 0.06 - 0.065)
4. Slice and view where model meets the interface later it adds the wrong color ABOVE the interface support which shows up attached to prints underside.

[image]

[image]

### Actual results

[image]

[image]

### Expected results

This should be expected results.

[image]

[image]

### Project file & Debug log uploads

[Orca Bug File.zip](https://github.com/user-attachments/files/19840066/Orca.Bug.File.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

### Anything else?

I will try to keep it brief as possible, this issue / bug has been an issue on bambu slicer for over 2 years now , I stayed with Orca cause this setting is detrimental to my business.  After downloading 2.3.0 this issue has now migrated to Orca. I hope this can be resolved before older versions get too far of a gap. I love Orca, its been truly a business saver for me. Thank you for taking the time to read! The proj file above just slice and check the first few layers you will notice the orange and black are under the yellow.

**Here is the bug report on the bambu github for lots of extra detail**
https://github.com/bambulab/BambuStudio/issues/3945

Noisyfox, 2025-04-22:
Will take a look soon.

CallmeBigfat, 2025-04-22:
> Will take a look soon.

Thank you, life saver.

Noisyfox, 2025-04-22:
Actually, in 2.2.0, the top z distance doesn't really works if it's set to smaller than half of the layer height, and with wipe tower enabled. It will use 0 instead. And as a workaround in 2.3 you could set this value to 0 and it should behave exactly as 2.2.

Noisyfox, 2025-04-22:
There will be very minor differences actually, but IMO should not make any difference in final print.

CallmeBigfat, 2025-04-22:
> There will be very minor differences actually, but IMO should not make any difference in final print.

Good find, it does seem to work if I were to print in 0.1 mm layer height and set the support layer z height to 0.06, those lines go away. I do know 0.09 makes the print really ugly at the bottom and actual 0 the supports took way too long to pull off. Why I found 0.06 - 0.08 the best. But if you say that value just defaults to 0 when the prime tower is enabled anyway then it may have been a placebo perhaps.

Noisyfox, 2025-04-22:
I haven't look at every part of the code yet, but if you set the distance to 0.06 and slice, and go to the gcode preview, you will see those support interfaces are still printed at the same nozzle z height as the object layers, which means it doesn't work as intended.

CallmeBigfat, 2025-04-22:
I was hoping I could use the height range modifier on the bottom few layers to "trick" it to working and print at 0.12 Layer height for the first few then go back to 0.18 (what I normally print these items at) but did not work. I wont lie I am still a bit confused but I want to thank you for looking into it so quickly, as you can see from the issue I linked its been quiet some time, so its nice to at least see why its doing this and how so.  I spent so many hours changing settings and I must have just never made the correlations that the support distance must be not lower than half of the print layers height. That at least gives me another step in the right direction.

CallmeBigfat, 2025-10-25:
still an issue :/