OrcaSlicer/OrcaSlicer #9444: Tree supports generated on angles over threshold

tflori, 2025-04-21:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

Windows

### OS Version

tested on Windows 11 / Linux

### Additional system information

tested on multiple systems with AMD CPUs

### Printer

Bambu Lab A1

### How to reproduce

1. open the 3mf file
2. slice the active plate

I'm trying to print KLP-Lampe keycaps on 106° rotation as I think that's the sweet spot but see self: 

### Actual results

It adds supports to the surface of the keys which are definetly not less than 20" as defined:

[image]

[image]

### Expected results

to add support only where needed. I found out that it is "better" when rotating the keycaps by 45° but this adds some "inaccuracy" that I don't want - also it adds only supports to one of the sides but the keycaps are symetrical and that does not make sense either

[image]

### Project file & Debug log uploads

[keycaps-print.zip](https://github.com/user-attachments/files/19840137/keycaps-print.zip)

[debug_Mon_Apr_21_22_33_09_21284.log.zip](https://github.com/user-attachments/files/19840439/debug_Mon_Apr_21_22_33_09_21284.log.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

discip, 2025-04-22:
Yes confirmed, same here.

@tflori
As a workaround:

[image]

[image]

discip, 2025-04-22:
Since most of them seem to be identical, you don't need to do every single one.
Just do the ones that are different, and then create the appropriate number of clones. 😊

tflori, 2025-04-22:
so duplicating copies the support information. good to know 👍

closed 2025-07-29 by github-actions[bot]