OrcaSlicer/OrcaSlicer #9446: Bottom layers not made correct in some situations in 2.3.1 but works in 2.3.0

Nukle0n, 2025-04-21:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.1 Beta ee7ebda

### Operating System (OS)

Windows

### OS Version

11 24h2

### Printer

Sovol SV06 Plus

### How to reproduce

This conical object which starts as two parts that merge has no bridging or solid internal layers in 2.3.1, but in 2.3.0 it does, same project.

### Actual results

[image]

### Expected results

[image]

### Project file & Debug log uploads

[top-red.zip](https://github.com/user-attachments/files/19840487/top-red.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

discip, 2025-04-22:
@Nukle0n
Would you mind sharing your project file instead of the **`STL`**?
Since it does work here. 😊

discip, 2025-04-22:
@Nukle0n
Would you mind sharing your project file instead of the **`STL`**?
Since it does work here. 😊

If I'm not mistaken, the only parameter you need to change is **`Ensure vertical shell thickness`**.
It needs to be set to **`All`**.

Nukle0n, 2025-04-22:
> [@Nukle0n](https://github.com/Nukle0n) Would you mind sharing your project file instead of the **`STL`**? Since it does work here. 😊
> 
> If I'm not mistaken, the only parameter you need to change is **`Ensure vertical shell thickness`**. It needs to be set to **`All`**.

[top-red (2).zip](https://github.com/user-attachments/files/19848034/top-red.2.zip)

Here's the project, sorry, it slipped my mind.  I tried changing `Ensure Vertical shell Thickness` to `All` but it doesn't change anything, Meanwhile if i slice it with the exact same settings as in 2.3.0 it does the bridging just fine.

discip, 2025-04-22:
> it does the bridging just fine.

So, your issue seems resolved.

If so, please don't forget to close this. 😊

Noisyfox, 2025-04-23:
Please give [#9332](https://github.com/SoftFever/OrcaSlicer/pull/9332) a try

Nukle0n, 2025-04-23:
> Please give [#9332](https://github.com/SoftFever/OrcaSlicer/pull/9332) a try

Hello, thanks for the reply, it seems this hasn't been merged yet? I'm sadly not equipped to build my own binaries, but I eagerly await it getting pushed into the "Nightly Builds". Do you want me to close it or should I wait and see if it works?

Noisyfox, 2025-04-23:
> Hello, thanks for the reply, it seems this hasn't been merged yet? I'm sadly not equipped to build my own binaries, but I eagerly await it getting pushed into the "Nightly Builds". Do you want me to [...]

You can download the PR build by going to the PR page, and there will be a "checks" tab on top of the page, where you can find an "Artifacts" drop down that you could download.

Nukle0n, 2025-04-23:
> You can download the PR build by going to the PR page, and there will be a "checks" tab on top of the page, where you can find an "Artifacts" drop down that you could download.

Ah, ok, yes I checked and it seems to slice ok in the windows portable artifact i downloaded from that pull

[image]

PR #9332 (2025-04-29): Merge tiny cracks that is part of a large bottom surface into bottom surface
Fix #9329
Fix #9446
closed 2025-04-23 by Nukle0n