OrcaSlicer/OrcaSlicer #9448: Tree support first layer not following speed rules set in slicer.

Mart7Mcfl7, 2025-04-22:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.1-dev

### Operating System (OS)

Windows

### OS Version

11

### Printer

Ender 3 - klipper

### How to reproduce

Slice model

### Actual results

Any tree support that starts on build plate will follow their own speed as set in splicer. This is far too fast for a first layer, causing failure.

### Expected results

First layer of supports should follow the 'first layer speed' section  

### Project file & Debug log uploads

[support bug_2.zip](https://github.com/user-attachments/files/19846985/support.bug_2.zip)

[debug_Mon_Apr_21_17_38_34_28024.log.zip](https://github.com/user-attachments/files/19847033/debug_Mon_Apr_21_17_38_34_28024.log.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Mart7Mcfl7, 2025-05-07:
Still happening after reverting to latest main - as you can see with speed settings the first layer of the supports should not be going at 150mm/s.

Unless I'm doing something wrong, the supports should follow first layer rules, like the blue sections of the first layer.

[image]

Mart7Mcfl7, 2025-05-21:
Still happening, no idea how to fix it. first layer should follow set speeds

[image]

coryrc, 2025-05-25:
For workaround, you should be able to use a single-layer-height modifier cube, where you change `Speed->Other layers speed->Support` to your desired slow speed.

coryrc, 2025-06-25:
My previous comment was wrong, because modifiers can only be applied per-object and supports are not part of the object, so a modifier cannot be used to adjust support first layers.

thebeardbe, 2025-09-20:
I have the same bug too. But I narrowed it down a bit more. If you use the organic style of tree the speed is actually respected if you use any other type the speed is not respected. 

Style: auto
[image]

Style: Organic
[image]

Style: Tree Slim
[image]

Style: Tree Strong
[image]

Style: Tree hybrid
[image]

I am on win10 using the orcaslicer 2.3.1-beta build 1e87666