OrcaSlicer/OrcaSlicer #9452: OrcaSlicer keeps crashing after starting a print

jkaczor, 2025-04-22:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

Windows

### OS Version

Windows 11 Pro (Build 26100)

### Additional system information

From Windows Applications event log:

Faulting application name: orca-slicer.exe, version: 1.0.0.0, time stamp: 0x67d99bcb
Faulting module name: jscript9Legacy.dll, version: 11.0.26100.3624, time stamp: 0x80e41c18
Exception code: 0xc000041d
Fault offset: 0x000000000013e11a
Faulting process id: 0x3B50
Faulting application start time: 0x1DBB3ABBFD0F802
Faulting application path: C:\Program Files\OrcaSlicer\orca-slicer.exe
Faulting module path: C:\Windows\System32\jscript9Legacy.dll
Report Id: f1c406d6-1c59-41a3-936d-50da63597592
Faulting package full name: 
Faulting package-relative application ID: 

### Printer

Bambu Lab X1C (X1Plus firmware)

### How to reproduce

Open OrcaSlicer, select a .3mf file - place, slicer and then print the model, go do other work - at some point OrcaSlicer has just dissapeared.

Note - I do not have a discrete graphics card, only basic Intel graphics on this machine.

### Actual results

OrcaSlicer is no longer loaded/running in memory.

### Expected results

OrcaSlicer should stay open until I close it ;-)

### Project file & Debug log uploads

[crash_Tue_Apr_22_13_43_59_0.zip](https://github.com/user-attachments/files/19854927/crash_Tue_Apr_22_13_43_59_0.zip)

[Unnamed-BodyPad001.zip](https://github.com/user-attachments/files/19854932/Unnamed-BodyPad001.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

jkaczor, 2025-04-22:
Update - looks like this only occurs if the Camera is live and enabled.

lotsahelp, 2025-04-23:
The last week Orca has been crashing every time I click print, immediately. I'm printing to a P1S with or without AMS doesn't seem to matter. I've used both an M4 and M2 mac. I'm not sure if the MacOS 15.4.1 update had anything to do with it as I wasn't paying  attention to that and I don't have a non-upgraded machine to test with. When reopening, it does bring my project back to where I left it. I also noticed that if I visit the device page first for the printer, then I can successfully print the file.

[orca crash.txt](https://github.com/user-attachments/files/19871004/orca.crash.txt)

nec3f, 2025-04-25:
"at some point OrcaSlicer has just dissapeared."
This is exactly what's been happening to me. I'll start a print, it might even be the second or third or 10th print, go back to what I was doing and a few minutes later I notice it's gone. 
I rarely use 3mf files, it happens all the time with STLs as well. 
I couldn't tell you if the camera is enabled or not when it happens.
It's been happening to me for the last several builds. 
It doesn't happen every time or even with any regularity. Sometimes it's happen with two or three prints in a row, sometimes it doesn't happen for a week. 

A new build came out right around the time the Bambu controversy was going on and I want to say that's when this started. However, that's also when I put my printer in LAN mode which caused a bunch of other problems/annoyances, and I may be grouping it in with that since it feels like it's been happening for longer than that. 

(Side note, I want to thank whoever is responsible for allowing it to go back to *exactly*, where you left off. Same settings, same layer height changes, same everything.   Come to think of it, I wonder if that means a 'restore previous session' feature could be enabled.)

HerrDrachen3D, 2025-05-05:
This happens to me as well - I can print 2,3,4,5 times with no issue and then I will print once more and Orca crashes.  Thankfully I can open it again and I can start from where I left off.  This has been happening for several versions of Orca but I just now got enough energy to report it.  Anything I can provide to be helpful?

mishak87, 2025-05-09:
It happens to me half of the time it is really annoying. All printers A1/P1S/A1m in LAN mode, cameras off. If I pick printer that is offline it crashes sometimes when it finally timeouts after a 30/60s of waiting.

rumangerst, 2025-05-09:
This also happened to me, but as I was unhappy with the P1S camera, I replaced the feed with a custom one provided by a PI with webcam. Did not crash again so far, so this seems to be related to Bambu cameras.

OS: Linux (CachyOS), AppImage

lin-ycv, 2025-06-25:
tryung to access the decive tab will cause orca to crash, after send a print, orca will crash (sometimes it'll crash before the job is succesfully sent)

I'm using 4 P1S all in lan mode (firmware version 01.08.01.00 (20250520)), with orca 2.3.0

```
Exceptions:
Exception Code :c0000005 ACCESS_VIOLATION
Exception Flag :0x0 NumberParameters :2 
Param 0 :0x0 
Param 1 :0x8 
Context :0000009146DFF340 
ContextFlag : 0x10005f, EFlags: 0x10202 
Fault address:  0xAA74914F 0x1:0x16814F C:\Users\User\AppData\Roaming\OrcaSlicer\plugins\bambu_networking.dll

Registers:
RAX:0000000000000000
RBX:00000208C6A9B170
RCX:00000208D3CC8208
RDX:00000208C6A9B170
RSI:00000208C6A9B178
RDI:0000009146DFFAF0
R8:7FFFFFFFFFFFFFFC
R9:0000000000000002
R10:00007FF9332E0000
R11:0000009146DFF980
R12:0000000000000000
R13:0000000000000000
R14:00000208D3CC8208
R15:0000000000000000
CS:RIP:0033:00007FF8AA74914F
SS:RSP:002B:0000009146DFFA50  RBP:00000000FFFFFFFF
DS:002B  ES:002B  FS:0053  GS:002B
Flags:00010202

Show CallStack:
7ff8aa74914f:bambu_network_user_logout [Unknown file][-1]
7ff8aa73067d:bambu_network_user_logout [Unknown file][-1]
7ff8aa72a75a:bambu_network_user_logout [Unknown file][-1]
7ff8aa72f3be:bambu_network_user_logout [Unknown file][-1]
7ff9370ee8d7:BaseThreadInitThunk [Unknown file][-1]
7ff9391dc34c:RtlUserThreadStart [Unknown file][-1]

Logical Address:
0xAA74914F 0x1:0x16814F C:\Users\User\AppData\Roaming\OrcaSlicer\plugins\bambu_networking.dll
0xAA73067D 0x1:0x14F67D C:\Users\User\AppData\Roaming\OrcaSlicer\plugins\bambu_networking.dll
0xAA72A75A 0x1:0x14975A C:\Users\User\AppData\Roaming\OrcaSlicer\plugins\bambu_networking.dll
0xAA72F3BE 0x1:0x14E3BE C:\Users\User\AppData\Roaming\OrcaSlicer\plugins\bambu_networking.dll
0x370EE8D7 0x1:0x2D8D7 C:\WINDOWS\System32\KERNEL32.DLL
0x391DC34C 0x1:0x3B34C C:\WINDOWS\SYSTEM32\ntdll.dll
```

mishak87, 2025-10-16:
Please isolate the plugin so it can be recovered/restarted without crashing the slicer.

kozmaz87, 2025-10-23:
Unstale it! This is happening to us all

kozmaz87, 2025-10-23:
> v2.3.0: This has been happening to me constantly the last week exactly as most are describing above. In my crashreports the offending culprit is usually wmp.dll but only once was it bambu_networking.dll. I haven't confirmed this yet but I have ben running two windows side by side (I have two X1C) and one will crash and the other seems to stay up. I don't usually leave the camera's running but obviously they are streaming. At first I thought it was because of the other bug I'm experiencing with plates disappearing on reload but now I'm beginning to wonder if it's just unstable. Seems like every time it crashes I lose all my saves (painting etc) even though I saved it multiple times already.
> 
> Attaching crash reports from orca logs and one of the windows error reporting outputs:
> 
> [crash_Fri_Jun_06_18_04_37_0.zip](https://github.com/user-attachments/files/20708979/crash_Fri_Jun_06_18_04_37_0.zip)
> 
> ```
> Faulting module name: wmp.dll, version: 12.0.26100.3912, time stamp: 0xeab032da
> Exception code: 0xc0000005
> Fault offset: 0x0000000000750680
> Faulting process id: 0xDDB4
> Faulting application start time: 0x1DBDA61A2DA0FA4
> Faulting application path: C:\Users\localadmin\bin\orca_slicer\orca-slicer.exe
> Faulting module path: C:\WINDOWS\system32\wmp.dll
> Report Id: c0fcc4f9-4982-4efa-b5e0-fc8aaf30db28
> Faulting package full name: 
> Faulting package-relative application ID:```
> ```

I have the exact same log in event viewer

smakd1, 2025-11-16:
I'm also having this issue. Here is the event log:

Log Name:      Application
Source:        Application Error
Date:          16/11/2025 7:38:54 PM
Event ID:      1000
Task Category: Application Crashing Events
Level:         Error
Keywords:      
User:          DESKTOP-O1Q8QA6\Ken-P5
Computer:      DESKTOP-O1Q8QA6
Description:
Faulting application name: orca-slicer.exe, version: 1.0.0.0, time stamp: 0x68dec417
Faulting module name: jscript9Legacy.dll, version: 11.0.26100.6725, time stamp: 0x6738c10a
Exception code: 0xc000041d
Fault offset: 0x000000000014482a
Faulting process id: 0x29D6C
Faulting application start time: 0x1DC568ACED8EF99
Faulting application path: C:\Program Files\OrcaSlicer\orca-slicer.exe
Faulting module path: C:\Windows\System32\jscript9Legacy.dll
Report Id: 08a087fb-e334-4f90-8ed0-a1fbc62601ab
Faulting package full name: 
Faulting package-relative application ID: 
Event Xml:
<Event xmlns="http://schemas.microsoft.com/win/2004/08/events/event">
  <System>
    <Provider Name="Application Error" Guid="{a0e9b465-b939-57d7-b27d-95d8e925ff57}" />
    <EventID>1000</EventID>
    <Version>0</Version>
    <Level>2</Level>
    <Task>100</Task>
    <Opcode>0</Opcode>
    <Keywords>0x8000000000000000</Keywords>
    <TimeCreated SystemTime="2025-11-16T09:38:54.6272311Z" />
    <EventRecordID>91619</EventRecordID>
    <Correlation />
    <Execution ProcessID="40516" ThreadID="122388" />
    <Channel>Application</Channel>
    <Computer>DESKTOP-O1Q8QA6</Computer>
    <Security UserID="S-1-5-21-2309628172-3646735362-1247960165-1001" />
  </System>
  <EventData>
    <Data Name="AppName">orca-slicer.exe</Data>
    <Data Name="AppVersion">1.0.0.0</Data>
    <Data Name="AppTimeStamp">68dec417</Data>
    <Data Name="ModuleName">jscript9Legacy.dll</Data>
    <Data Name="ModuleVersion">11.0.26100.6725</Data>
    <Data Name="ModuleTimeStamp">6738c10a</Data>
    <Data Name="ExceptionCode">c000041d</Data>
    <Data Name="FaultingOffset">000000000014482a</Data>
    <Data Name="ProcessId">0x29d6c</Data>
    <Data Name="ProcessCreationTime">0x1dc568aced8ef99</Data>
    <Data Name="AppPath">C:\Program Files\OrcaSlicer\orca-slicer.exe</Data>
    <Data Name="ModulePath">C:\Windows\System32\jscript9Legacy.dll</Data>
    <Data Name="IntegratorReportId">08a087fb-e334-4f90-8ed0-a1fbc62601ab</Data>
    <Data Name="PackageFullName">
    </Data>
    <Data Name="PackageRelativeAppId">
    </Data>
  </EventData>
</Event>

kazolar, 2025-12-27:
This is super consistent. Lan mode, video feed enabled -- start print -- crashes.

lotsahelp, 2026-01-28:
I continue getting crashes any time the first printer in my list is offline.