OrcaSlicer/OrcaSlicer #9457: time remaining is inacurate and way off

kbuggenhout, 2025-04-23:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

Windows

### OS Version

win 11

### Printer

AnkerMake M5C

### How to reproduce

1. load any STL file, 
2. slice
3. note time estimate to completion from slicer
4. send to printer
5. check time remaining in printer console/progress app 

### Actual results

a simple 50min print according to the slicer, is reported to take between 58 and 73 hours to complete. actual print time : ~50mins

### Expected results

closer to accurate time remaining used in g-code

### Project file & Debug log uploads

none

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

checked in prusa slicer and an older build of orca ( < 2.2) which do not show the exagerated time remaining
so this is a recent change in orca slicer that broke time remaining being reported correctly

igiannakas, 2025-04-23:
I've seen something similar. Specifically the inner perimeters are the ones driving the majority of the deviation, especially in "organic" models with curves etc.

My current hypothesis is that this is an issue due to the estimator not taking into account the "exit" velocity from those micro segments and artificially believing that the speed is slower than it actually is.

@Noisyfox @SoftFever 

I've been trying to figure this out for the past week but with limited success. Would love your thoughts.

See below issue specifics - internal perimeter accel = 10k
Inner wall time estimate = 1h 4m
[image]

Increasing speed of inner walls to 300mm/sec
Inner wall time estimate = 1h 4m
[image]

**internal perimeter accel = 3k**
Inner wall time estimate = 1h 11m
[image]

Increasing speed of inner walls to 300mm/sec
Inner wall time estimate = 1h 51m! Its actually gone up even though the speed is higher!
[image]

**Enabling arc fitting:**

[image]
Increasing speed of inner walls to 300mm/sec
[image]

**My gut is telling me that the line segments present when printing circular objects, even though they dont materially slow down the printer in reality, are counted as "stoppages" or materially affect the "exit velocity" used in the trapezoid calculation to estimate print time.** Hence why when using arc fitting the estimates are more in line with what I'd expect them to be when increasing print velocity.

igiannakas, 2025-04-23:
BTW this issue was present in 2.2 as well but not as pronounced. The estimates were still off but now they are even more so.

igiannakas, 2025-04-23:
Take a look here: https://github.com/SoftFever/OrcaSlicer/commit/14d49b738fe5708394e2d02b732c4ec667a7f26f

and here: https://github.com/SoftFever/OrcaSlicer/commit/ab7ef4f55efeabd92b5b303fd4be07b522af3cc5

I think the slicer time estimates are way off too. I dont think there is good use of the klipper SCV values to accurately estimate the exit velocity in circular objects. Try putting a bed full of cylinders and try increasing the print speed. the time actually goes up, even if you set a crazy high flow rate etc.

kbuggenhout, 2025-04-23:
seems that there was a duplicate function causing issues, but removing the postprocessing part handling M73 messages is now totally borked. added some comments to both merged issues above, that this caused further regression with certain printers

kbuggenhout, 2025-04-23:
currently I see only 1 difference in the generated G-code, with prusa-slicer / orca-slicer and eufymake slicer. in the config section of the gcode at the end, orca gcode does not have a line : ";remaining_times=1" , other slicers and a pre 2,3 orca generate this line.
which then result in a correctly displayed remaining time.  

also quite curious if other brands that are Marlin(2) based printers show the same issue : highly exagerated remaining times in the 100x to 500x range.

kbuggenhout, 2025-04-24:
ran a few more tests and found something quite odd : a simple flat 20x20x1 mm object times are acurate ( there are some differences in profiles that account for the minor difference.

orca slicer time estimate 2m 
[image]
prusa slicer time estimate 1m42 
[image]

but!!!
when I slice an object that is bigger like a 20x20x20 mm cube... orca times are suddenly 7-8hours while prusa is quite consistent in showing remaining time to be ~13mins, seems that there is a multiplication done per layer. while I have to dig into the g-code itself to find what is the difference in generation, and which gcode command causes the offset.

inserting the ; time-remains=1 into the gcode does nothing as far as I can tell.

kbuggenhout, 2025-04-24:
redid a 20x20x3mm and time is gone crazy again : 34h up to 78hours for a slicer estimated 3minutes print

[image]

[image]

kbuggenhout, 2025-04-24:
some more depth here, I dont know where this value comes from but snooping on the MQTT messages between the printer and the app, I see numbers that are not corresponding.

there is a TIME string in seconds that from Orca slicer are wildly varying. 

sample from orca sliced print
MQTT message: {'commandType': 1001, 'time': 208441, 'totalTime': 24, 
MQTT message: {'commandType': 1001, 'time': 221970, 'totalTime': 128,
MQTT message: {'commandType': 1001, 'time': 162942, 'totalTime': 183,

sample from prusa sliced print
MQTT message: {'commandType': 1001, 'time': 216, 'totalTime': 48, 
MQTT message: {'commandType': 1001, 'time': 216, 'totalTime': 154,

difference here is that time is fixed when prusa/AM/EM slicer is used, while Orca Time value moves in all directions and is off... when I divide the value from prusa by 60, I get the estimated time as being 3m30s which is the slicers estimate for the print. while dividing the orca reported time value is indeed fluctuating between 50h and 34hours between steps... this is definitely not using the M73 R value , which is being posted as R3 at percentage 0  ( 3 mins at start)

that time value is populated by something in the gcode, but I am at a loss what this is.

kbuggenhout, 2025-06-23:
apparently someone found a potential fix for this, will see if I can create a custom ( printer dependant ) have to find a way to insert TIME with the estimated time in seconds

https://www.reddit.com/r/AnkerMake/comments/1l7vcmy/tip_when_using_orcaslicer_you_can_manually_add/

kbuggenhout, 2025-06-23:
amker printer profiles are missing the following command in machine G-code below G28 ; Home

;LAYER_COUNT:{total_layer_count}

wil test tomorow and do a github pull request if positive. change will only apply to anker M5/M5C printers