OrcaSlicer/OrcaSlicer #9460: Purge Line is a Different Filament

aBusto3, 2025-04-23:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Prusa XL 0.4mm Single Tool

### How to reproduce

1. Set printer to Prusa XL 0.4 nozzle default preset
2. Add a second filament 
3. Add an object to the build plate
4. Set the object as the 2nd filament listed
5. Slice

### Actual results

The object is set as the desired filament, but the prime line is always the first filament in the list. When loaded to the printer is tries to map filament to different tools when it is a single tool

### Expected results

The prime line should be the same filament as the objects in the build plate since this is a single tool printer.

### Project file & Debug log uploads

[WrongPrimeLine.zip](https://github.com/user-attachments/files/19873215/WrongPrimeLine.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

aBusto3, 2025-04-23:
[image]

Steiner-SE, 2025-04-25:
Yes, I have the same issue, even if the #1 color isn't used at all it will still load, purge, then instantly go to filament change, purge again and then start printing. Very wasteful and time consuming. Tried the color priority customization, and it works for prints, but doesn't affect the purge line, that's ALWAYS color #1.
You'd expect the first color used ina  print, or if one color print the used color to be purged, not a color that isn't in use.