OrcaSlicer/OrcaSlicer #9472: T0 Code generated in single extruder setup

tworkman08, 2025-04-24:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Elegoo Neptune 4 Pro

### How to reproduce

Create Project,
Add 2 filaments,
Slice Project with two materials used,
Export to G-Code

### Actual results

Initial Filament and changes generate T0,T1, etc despite only one extruder and no AMS system of anytype configured. 

### Expected results

Accepted practice in a single extruder, non-ams klipper is to use M600. T codes are ge

[tcode_issue.zip](https://github.com/user-attachments/files/19899066/tcode_issue.zip)

nerally a toolhead change, which is not expected in this setup. Post processing to remove T codes or creating a custom macro so klipper doesn't throw an error seems to be less logical than generating T codes without having configured anything to prompt that.

### Project file & Debug log uploads

[tcode_issue.zip](https://github.com/user-attachments/files/19899079/tcode_issue.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

tlhintoq, 2025-04-26:
Just another user like you (not dev team).
But you have my curiosity up.

> Elegoo Neptune 4 Pro
How to reproduce
Create Project,
**_Add 2 filaments,
Slice Project with two materials used,_**
Export to G-Code

Q1: How are you doing 2 materials if you're on a neptune4?
Q2: If you are doing two materials, then a material/tool change sounds right.  Its up to your machine to know if that's changing entire tool heads, if its a 2-in-1 head, etc.

If you're doing manual material changes, maybe you just want to `pause` instead of slicing for 2 materials?

tworkman08, 2025-04-26:
> Just another user like you (not dev team). But you have my curiosity up.
>
>
> Q1: How are you doing 2 materials if you're on a neptune4? Q2: If you are doing two materials, then a material/tool change [...]

This is an example project for this bug. Manual material changes, yes. Single extruder with M600(pause) configured in orca for filament changes. I slice with multiple materials precisely so that orca inserts the pauses from painting before slicing and also because on occasion I need different filament profiles within the object - which I don’t believe is possible with simply inputting a pause. As described the options currently are to configure klipper so it ignores T commands or post process the gcode to remove it.

tlhintoq, 2025-04-27:
Makes sense.  Cool.
Notepad++ would make fast work of the `T0` post processing.

goodjaerb, 2025-04-27:
Forgive me if I'm out of line but do you have, under Printer Setting/Multimaterial, both the Single Extruder Multi Material AND the Manual Filament Change checkbox enabled? The second check box tooltip states it removes the T0 commands. I have used this with no issue.

tworkman08, 2025-04-27:
> Forgive me if I'm out of line but do you have, under Printer Setting/Multimaterial, both the Single Extruder Multi Material AND the Manual Filament Change checkbox enabled? The second check box tooltip states it removes the T0 commands. I have used this with no issue.

Not out of line at all, that actually appears to be exactly what had somehow changed in my config. it coincided with some updates and multiple searches yielded that you couldn't omit with a setting. thanks for your input, ill close this bug.

closed 2025-04-27 by tworkman08

is-andrade, 2025-06-15:
I Solved it by going in Printer config > Multimaterial and check the "Manual Filament Change"
now intead of a `T0` in the gcode I see: `; MANUAL_TOOL_CHANGE T0`