OrcaSlicer/OrcaSlicer #9474: Using Zink on Nvidia only Model and Preview panels work

DaRacci, 2025-04-25:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0-rc

### Operating System (OS)

Linux

### OS Version

NixOS Unstable

### Additional system information

<details>
<summary>fastfetch</summary>

[image]

</details>

<details>
<summary>nix-info</summary>

 - system: `"x86_64-linux"`
 - host os: `Linux 6.13.7-xanmod1, NixOS, 25.05 (Warbler), 25.05.20250319.a84ebe2`
 - multi-user?: `yes`
 - sandbox: `yes`
 - version: `nix-env (Lix, like Nix) 2.91.1
System type: x86_64-linux
Additional system types: i686-linux, x86_64-v1-linux, x86_64-v2-linux, x86_64-v3-linux, x86_64-v4-linux
Features: gc, signed-caches
System configuration file: /etc/nix/nix.conf
User configuration files: /home/racci/.config/nix/nix.conf:/nix/store/1nmjv4g3sf1jr8h2lfy36i6d0fpza1iv-stylix-kde-config/nix/nix.conf:/etc/xdg/nix/nix.conf:/home/racci/.nix-profile/etc/xdg/nix/nix.conf:/nix/profile/etc/xdg/nix/nix.conf:/home/racci/.local/state/nix/profile/etc/xdg/nix/nix.conf:/etc/profiles/per-user/racci/etc/xdg/nix/nix.conf:/nix/var/nix/profiles/default/etc/xdg/nix/nix.conf:/run/current-system/sw/etc/xdg/nix/nix.conf
Store directory: /nix/store
State directory: /nix/var/nix
Data directory: /nix/store/riqr33jgc60nsk8w2d0blq17y8skp0yn-lix-2.91.1/share`
 - nixpkgs: `/nix/store/1728d3jg85mkz2w2cvk6vi74i30fn6x7-source`
</details>

<details>
<summary>glxinfo</summary>

name of display: :0
display: :0  screen: 0
direct rendering: Yes
Memory info (GL_NVX_gpu_memory_info):
    Dedicated video memory: 24576 MB
    Total available memory: 24576 MB
    Currently available dedicated video memory: 21500 MB
OpenGL vendor string: NVIDIA Corporation
OpenGL renderer string: NVIDIA GeForce RTX 3090/PCIe/SSE2
OpenGL core profile version string: 4.6.0 NVIDIA 570.124.04
OpenGL core profile shading language version string: 4.60 NVIDIA
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile

OpenGL version string: 4.6.0 NVIDIA 570.124.04
OpenGL shading language version string: 4.60 NVIDIA
OpenGL context flags: (none)
OpenGL profile mask: (none)

OpenGL ES profile version string: OpenGL ES 3.2 NVIDIA 570.124.04
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20
</details>

<details>
<summary>vulkaninfo</summary>
❯ nix shell nixpkgs#vulkan-tools --command vulkaninfo --summary
WARNING: [Loader Message] Code 0 : terminator_CreateInstance: Received return code -3 from call to vkCreateInstance in ICD /
==========
VULKANINFO
==========

Vulkan Instance Version: 1.4.304

Instance Extensions: count = 25
-------------------------------
VK_EXT_acquire_drm_display             : extension revision 1
VK_EXT_acquire_xlib_display            : extension revision 1
VK_EXT_debug_report                    : extension revision 10
VK_EXT_debug_utils                     : extension revision 2
VK_EXT_direct_mode_display             : extension revision 1
VK_EXT_display_surface_counter         : extension revision 1
VK_EXT_headless_surface                : extension revision 1
VK_EXT_surface_maintenance1            : extension revision 1
VK_EXT_swapchain_colorspace            : extension revision 5
VK_KHR_device_group_creation           : extension revision 1
VK_KHR_display                         : extension revision 23
VK_KHR_external_fence_capabilities     : extension revision 1
VK_KHR_external_memory_capabilities    : extension revision 1
VK_KHR_external_semaphore_capabilities : extension revision 1
VK_KHR_get_display_properties2         : extension revision 1
VK_KHR_get_physical_device_properties2 : extension revision 2
VK_KHR_get_surface_capabilities2       : extension revision 1
VK_KHR_portability_enumeration         : extension revision 1
VK_KHR_surface                         : extension revision 25
VK_KHR_surface_protected_capabilities  : extension revision 1
VK_KHR_wayland_surface                 : extension revision 6
VK_KHR_xcb_surface                     : extension revision 6
VK_KHR_xlib_surface                    : extension revision 6
VK_LUNARG_direct_driver_loading        : extension revision 1
VK_NV_display_stereo                   : extension revision 1

Instance Layers: count = 14
---------------------------
VK_LAYER_ALVR_capture               ALVR display intercept layer                   1.0.68   version 1
VK_LAYER_FROG_gamescope_wsi_x86_64  Gamescope WSI (XWayland Bypass) Layer (x86_64) 1.3.221  version 1
VK_LAYER_INTEL_nullhw               INTEL NULL HW                                  1.1.73   version 1
VK_LAYER_MANGOHUD_overlay_32_x86    Vulkan Hud Overlay                             1.3.0    version 1
VK_LAYER_MANGOHUD_overlay_64_x86_64 Vulkan Hud Overlay                             1.3.0    version 1
VK_LAYER_MESA_device_select         Linux device selection layer                   1.4.303  version 1
VK_LAYER_MESA_overlay               Mesa Overlay layer                             1.4.303  version 1
VK_LAYER_NV_optimus                 NVIDIA Optimus layer                           1.4.303  version 1
VK_LAYER_OBS_vkcapture_32           OBS Linux game capture                         1.4.304  version 1
VK_LAYER_OBS_vkcapture_64           OBS Linux game capture                         1.4.304  version 1
VK_LAYER_VALVE_steam_fossilize_32   Steam Pipeline Caching Layer                   1.3.207  version 1
VK_LAYER_VALVE_steam_fossilize_64   Steam Pipeline Caching Layer                   1.3.207  version 1
VK_LAYER_VALVE_steam_overlay_32     Steam Overlay Layer                            1.3.207  version 1
VK_LAYER_VALVE_steam_overlay_64     Steam Overlay Layer                            1.3.207  version 1

Devices:
========
GPU0:
        apiVersion         = 1.4.303
        driverVersion      = 570.124.4.0
        vendorID           = 0x10de
        deviceID           = 0x2204
        deviceType         = PHYSICAL_DEVICE_TYPE_DISCRETE_GPU
        deviceName         = NVIDIA GeForce RTX 3090
        driverID           = DRIVER_ID_NVIDIA_PROPRIETARY
        driverName         = NVIDIA
        driverInfo         = 570.124.04
        conformanceVersion = 1.4.1.0
        deviceUUID         = cfa6b6de-abbe-2350-b1cc-17d4e608fb19
        driverUUID         = 4fef4b44-6f8c-56b2-9b76-6a75892bbaa4
GPU1:
        apiVersion         = 1.4.305
        driverVersion      = 25.0.1
        vendorID           = 0x1002
        deviceID           = 0x13c0
        deviceType         = PHYSICAL_DEVICE_TYPE_INTEGRATED_GPU
        deviceName         = AMD Radeon Graphics (RADV RAPHAEL_MENDOCINO)
        driverID           = DRIVER_ID_MESA_RADV
        driverName         = radv
        driverInfo         = Mesa 25.0.1
        conformanceVersion = 0.0.0.0
        deviceUUID         = 00000000-7400-0000-0000-000000000000
        driverUUID         = 414d442d-4d45-5341-2d44-525600000000
GPU2:
        apiVersion         = 1.4.305
        driverVersion      = 0.0.1
        vendorID           = 0x10005
        deviceID           = 0x0000
        deviceType         = PHYSICAL_DEVICE_TYPE_CPU
        deviceName         = llvmpipe (LLVM 19.1.7, 256 bits)
        driverID           = DRIVER_ID_MESA_LLVMPIPE
        driverName         = llvmpipe
        driverInfo         = Mesa 25.0.1 (LLVM 19.1.7)
        conformanceVersion = 1.3.1.1
        deviceUUID         = 6d657361-3235-2e30-2e31-000000000000
        driverUUID         = 6c6c766d-7069-7065-5555-494400000000
<details>

### Printer

Creality K1C

### How to reproduce

* Run `__GLX_VENDOR_LIBRARY_NAME=mesa MESA_LOADER_DRIVER_OVERRIDE=zink GALLIUM_DRIVER=zink WEBKIT_FORCE_COMPOSITING_MODE=0 WEBKIT_DISABLE_COMPOSITING_MODE=0 __EGL_VENDOR_LIBRARY_FILENAMES=/nix/store/2h2liykjl4vx144vmgm5m993n1liqs62-graphics-drivers/share/glvnd/egl_vendor.d/50_mesa.json nix run nixpkgs#orca-slicer --impure` to use zink
* Try to open the Home, Device or Project Page and be greeted by a blank screen.
  * Additionally menus such as trying to add a new printer / filament preset are also blank.

### Actual results

The pages and popups are shown as completed blank pages.

Additionally when opening the popups like adding a new filament preset the logs display this.
```
** (orca-slicer:420292): CRITICAL **: 15:24:39.493: Cannot register URI scheme wxfs more than once
** (orca-slicer:420292): CRITICAL **: 15:24:39.493: Cannot register URI scheme memory more than once
Could not create GBM EGL display: EGL_NOT_INITIALIZED. Aborting...
```

<details>
<summary>Screenshots</summary>

[image]

[image]

<details>

### Expected results

Not rendered blank.

### Project file & Debug log uploads

[debug_Fri_Apr_25_15_29_33_437169.log.zip](https://github.com/user-attachments/files/19903352/debug_Fri_Apr_25_15_29_33_437169.log.zip)

### Checklist of files to include

- [x] Log file
- [ ] Project file

XperiorTR, 2025-05-10:
i'm also having this issue, the previous workaround of using software rendering via `__GLX_VENDOR_LIBRARY_NAME=mesa __EGL_VENDOR_LIBRARY_FILENAMES=/run/opengl-driver/share/glvnd/egl_vendor.d/50_mesa.json orca-slicer
` also seems to be broken in the same way
note that im on nixos, i wonder if this is a nix package issue

DaRacci, 2025-06-07:
I'm really confused on whats going on but I've just gotten this to work by setting only `GBM_BACKEND=dri` as the only environment variable before launching, in the console its complaining a lot but GPU acceleration seems to be working and no menus are blank, the message in the console is `MESA: error: zink: failed drmPrimeFDToHandle Invalid argument` and seems to be printing whenever it would have instead triggered the `EGL_NOT_INITIALIZED` error.

closed 2025-09-14 by github-actions[bot]