OrcaSlicer/OrcaSlicer #9475: flow calibration ugly text slicing

jlo1975, 2025-04-25:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

Hello,
with new flow calibration pattern using the concentric/archi cords surface pattern.
The first layer is no more plain uniform and contains holes following the text pattern.
But this holes are not fully clean (see screenshot). 
Leading to very ugly and unclear text rendering.

first layer

[image]

second layer
[image]

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

maybe ensure the first layer is fully uniform without the text ''holes'' like on previous version of flow calibration pattern.

Noisyfox, 2025-04-25:
This should be fixed by #9332, please give it a try:
[image]

jlo1975, 2025-04-27:
By reading the dicussion you have on the merge tiny cracks topic.
Another way to easely fix the flow calibration test would be to ensure the text start at layer 2 and not layer 1 (based on 0.2  layer height). Maybe use the same embossing as the one on PA calibration.

closed 2025-04-29 by SoftFever