OrcaSlicer/OrcaSlicer #9477: Confusing Settings Dialog

klwemu, 2025-04-25:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

This dialog is not intuitive at all! 
I always have to thing what button does what action. 

[image]

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

From my point of view it is not enough the change the button labels, it needs a complete redesigned.

MxBrnr, 2025-04-25:
Duplicate of https://github.com/SoftFever/OrcaSlicer/issues/4422. Please make sure you are checking for existing issues before opening a new one, especially since you indicated that you did this.

nilesjohnson, 2025-05-02:
#4422 is closed; is it going to be reopened?

MxBrnr, 2025-05-03:
> [#4422](https://github.com/SoftFever/OrcaSlicer/issues/4422) is closed; is it going to be reopened?

Nope. The developers have already said in that closed issue that they will not be changing it.