OrcaSlicer/OrcaSlicer #9478: the ADAPTIVE button does not work

Yevgeniy-Gonzha, 2025-04-25:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0 (Build 70931e5)

### Operating System (OS)

Windows

### OS Version

11

### Additional system information

CPU: Intel Core i7-10700K CPU @ 3.80GHz
RAM: 32 GB
NVIDIA GeForce RTX 3060

[image]

### Printer

BambuLab P1S with AMS

### How to reproduce

1. 

### Actual results

1. choose a model
2. click the Adaptive Layers button on the toolbar
3. in the drop-down panel, click the “Adaptive” button
4. nothing happens (you can see it on the vertical layer indicator on the right)

BUT
When I manually change the height of the layers, the layer height indicator (on the right) shows it.

### Expected results

the height of the layers should have been adapted
but this did not happen

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/19908625/log.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

Evixone, 2025-04-27:
Same for me

BodyBas, 2025-05-10:
Same here, it works as preview in "prepare" tab, but not present after slicing a model (same version and build).

closed 2025-08-17 by github-actions[bot]