OrcaSlicer/OrcaSlicer #9480: Slow Down for Curled Perimeters / Overhangs introduces artifacts and reduces quality

ShatterSide, 2025-04-25:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

Windows

### OS Version

Windows 10

### Printer

Bambu Lab X1C with E3D Obxidian hotend, otherwise stock machine

### How to reproduce

1. Load Benchy model.
2. Use any standard filament profile, (PLA will do).
3. Use 0.2mm Standard preset.
4. Slice and view preview by Speed or Flow.
5. Observe the choppy, stepped, speed changes of the front hull, which translate into artifacts in the print itself.
6. Turn off Slow Down for Curled Perimeters and re-slice.
7. Observe the choppy, stepped, speed changes are highly reduced, which also correlates with the print itself.
8. Turn off Slow Down for Overhangs.
9. Observe the preview is now smooth, which follows through with the print itself.

### Actual results

[image]

[image]

[image]

### Expected results

A much smoother speed transition. The following screenshot is from default Bambu Studio profile.

[image]

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

Possible discussion points and suggestions.

1. Disable "Slow down for curled perimeters" BY DEFAULT.
2. Artifacting could be caused by the stepped speed values, and micro accelerations between them?
3. Could be as overhang changes around compound curves exacerbates the issue.

No artifacting in Bambu Slicer. This is almost certainly some function of "Slow down for curled perimeters", and to a lesser extend, "Slow down for overhangs".

igiannakas, 2025-04-29:
This is due to PA tuning being a bit off and due to the lowest overhang value being too slow, hence the speed delta being too large for the extruder to handle it

Solution is to calibrate the printer's PA / K value a bit more accurately and to increase the minimum print speed up from 10mm/sec to something closer to 30-40mm/sec (as 10mm/sec is unnecessarily slow).

You can see how it works when this is setup correctly here: https://github.com/SoftFever/OrcaSlicer/issues/9424#issuecomment-2836082176

You can also enable extrusion rate smoothing (only on external perimeters) to allow the printer to better match the extruder speed to the actual mechanical limits of the machine. You can see print test here: https://github.com/SoftFever/OrcaSlicer/issues/9424#issuecomment-2836225304

ShatterSide, 2025-04-29:
> This is due to PA tuning being a bit off and due to the lowest overhang value being too slow, hence the speed delta being too large for the extruder to handle it. Basically it is over extruding when moving slower and during the transition from fast to slow and slow to fast.
> 

I am fastidious with my PA tuning. My filaments are calibrated as perfect as can be. I manually calibrate since I can get better results manually than the LiDAR can.

> Solution is to calibrate the printer's PA / K value a bit more accurately (run the test with 0.001 resolution) and to increase the minimum print speed up from 10mm/sec to something closer to 30-40mm/sec (as 10mm/sec is unnecessarily slow) to reduce the speed delta.
> 

I run the test with 0.0005 resolution. I do both the tower AND the pattern for every filament. I am anal about this.

My minimum speed is 20mm/s for most filaments anyway. Even if I increase it to 100mm/s it doesn't change the stepped speed changes in the preview, or it changes for the worse, since now the differential is larger.

[image]

> You can see how it works when this is setup correctly here: [#9424 (comment)](https://github.com/SoftFever/OrcaSlicer/issues/9424#issuecomment-2836082176)
> 
> You can also enable extrusion rate smoothing (only on external perimeters) to allow the printer to better match the extruder speed to the actual mechanical limits of the machine. You can see print test here: [#9424 (comment)](https://github.com/SoftFever/OrcaSlicer/issues/9424#issuecomment-2836225304)

I can try Extrusion rate smoothing for a print, but the preview of course does not get better. I am using an X1C and tried values of 300 - 350 mm3/s2.

This is trying multiple filament profiles with and without PA enabled. All are calibrated perfectly.

I didn't have these artifacts a few versions of Orca ago, and I don't get them with Bambu Studio either.

I am quite certain that the artifacts here are a result of this stepped speed change instead of a smooth transition. Of course PA is going to be tested during this, but if extrusion rate smoothing is the band-aid fix, then perhaps that feature should be enabled by default as long as Slow down for curled perimeters is default.

Otherwise, surely the Slow down for curled perimeters could do some interpolation?

igiannakas, 2025-04-29:
You can see the effect of extrusion rate smoothing here: https://github.com/SoftFever/OrcaSlicer/pull/2161#issuecomment-1732577104

That was on my X1C

This heavily depends on the printer. For example my voron doesn’t need it at all vs the X1C that needed a bit of help, mostly due to the fast print accelerations putting a high load on the extruder when rapidly changing speeds within the same feature. Same as bridging artefacts (ie under extrusion after a bridge) that are common on the X1C.

A lower value than 300 is needed. ERS was reworked to be less sensitive. Segment length of 0.5/1mm and value of 60-100 is more approrpiate now. I need to update the tooltip.

In general you want half a cm or a cm of speed gradient to eliminate this.

Bambu slicer has also introduced something similar (speed gradients) for overhangs, that is also activated separately.

igiannakas, 2025-04-29:
However ERS disables arc fitting.

So there is a tradeoff. In my tuned X1C profiles I have reduced wall accelerations down to 2k and max speed down to 180mm/sec with min slowdown down to 30-40mm/sec. This combats this issue for the vast majority of cases. Edge cases I deal with ERS or when I want extra high quality perimeters by dropping external perimeter speed further.

staal54a, 2025-07-20:
@igiannakas I was looking through various issues here as I was having some underextrusion problems after overhangs on my X1C with the old ERS values and you made a few points that stuck out to me:

1. Along with the tooltip, the [ERS wiki page](https://github.com/SoftFever/OrcaSlicer/wiki/speed_settings_advanced) may need to be updated with the new, lower values. I assume that is planned at some point, but in the meantime, is there quick guidance on what some suggested values would be appropriate for different ranges of flow, acceleration and layer height/width? I have an X1C so I can use the 60-100 you mentioned here (which seems to have fixed my issue) but I also have other printers (bedslingers) with varying max speeds/accelerations so guidance for those would be helpful.
2. Are Bambu machines not able to make use of the adaptive pressure advance feature in OrcaSlicer? I haven't really played around with it but it never occurred to me that it might not work for my X1C (I do use it on my other machines though).
3. A little less relevant, but do you have your tuned X1C profiles uploaded somewhere where they can be viewed? I usually do a good job of tuning in my profiles but I always go back and find another setting or two that I can optimized a bit more so it'd be great to compare and see what a different profile uses.

Thanks!

igiannakas, 2025-07-20:
Hi there!! 

Indeed the tooltip needs updating… Will get to it at some point :)

Adaptive PA during testing with the x1 didn’t work well for a number of users hence not really recommended. But feel free to try and report back. It won’t break your machine but it may not be delivering the expected benefit.

I’ve got my profiles on the orca discord here: https://discord.com/channels/1137181739773603922/1188430865005551656

Inanias, 2026-04-17:
So let me get this straight: you have an X1C, you KNOW that this does not work out of the box and still you insist that this is on by default?
Are these defaults only for people who tune their printer to the linking of the developers?

Pretty sure safe/sane defaults are something else.