OrcaSlicer/OrcaSlicer #9488: Orcaslicer appears to have a problem with  Nvidia Quadro RTX-A4000 (nvoglv64.dll) Crashes when slicing

W1SPS, 2025-04-26:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.1-DEV

### Operating System (OS)

Windows

### OS Version

Windows 11

### Additional system information

MSI Z690 Tomahawk WIFI 3.0
Intel Core I9-12900K
Corsair 4x32 DDR5-5200 (64 GB)
Nvidia Quadro RTX-A4000
(2) Samsung SSD980Pro 1TB
Windows 11

### Printer

Bambu X1C

### How to reproduce

Slice anything over a certain size or number of parts, Orcaslicer appears to have a problem with  Nvidia Quadro RTX-A4000 (nvoglv64.dll) Crashes when finishing slicing. It has been doing this back prior to 2.3.0. I am running the newest version of Nvidia's driver Release 570 version 572.83.
This is a brand new re-install of windows and all of my base software. This problem existed before and is still a problem.

### Actual results

Program simply halts and shuts down

### Expected results

Slice should complete and program shouldn't abort

### Project file & Debug log uploads

[crash_Sat_Apr_26_10_40_55_0.zip](https://github.com/user-attachments/files/19923056/crash_Sat_Apr_26_10_40_55_0.zip)

[Several other crash logs.zip](https://github.com/user-attachments/files/19923059/Several.other.crash.logs.zip)

[4 Complete 22 Cal Magazine-PETG-CF-GREEN.zip](https://github.com/user-attachments/files/19923066/4.Complete.22.Cal.Magazine-PETG-CF-GREEN.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

Noisyfox, 2025-04-26:
Dup #9381

closed 2025-04-26 by Noisyfox

Noisyfox, 2025-04-27:
Btw, does this works for you? https://forum.bambulab.com/t/bambu-studio-crashes-after-slicing-solved-nvidia-problem/162392