OrcaSlicer/OrcaSlicer #9492: Walls not being printed

moth4017, 2025-04-27:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

Windows

### OS Version

win 10 pro

### Additional system information

[image]

### Printer

Valkyrie HT 

### How to reproduce

I have a model with the  wall at 0.66mm wide 

[image]
 if i use clastic wall generation the wall is ignored  when my line width is set to 0.66mm

[image]

if i use Arachne 
the wall is printed at the correct wall thinkness 0.66

[image]

 

### Actual results

as above

### Expected results

classic to print wall at  0.66mm

### Project file & Debug log uploads

[orca_dev MSM_testprint v15.zip](https://github.com/user-attachments/files/19925999/orca_dev.MSM_testprint.v15.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

igiannakas, 2025-04-30:
Line width is not equal to wall width. Reduce your line width a little.

More details here: https://manual.slic3r.org/advanced/flow-math

moth4017, 2025-07-30:
issues still there even with the nightly builds