OrcaSlicer/OrcaSlicer #9505: Allow wipe before external loop on only one wall

vovodroid, 2025-04-28:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

Currently wipe before external loop works when there is internal wall.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Comment in GCode.cpp  says

`If region perimeters size not greater than or equal to 2, then skip the wipe inside move as we will extrude in mid air as no neighbouring perimeter exists.`

But what's wrong with such extrusion? I recently printed one wall TPU model and lacking of wipe produced blobs on unretruction.

igiannakas, 2025-04-30:
its because the de-retraction would happen on one side of the wall, instead of on the wall, leading to a blob on the perimeter.https://github.com/SoftFever/OrcaSlicer/pull/3616

vovodroid, 2025-04-30:
Original object had "double sided" external wall (i.e. external from both sides):

[image]

And I mean "single sided "external:

[image]

So deretraction could be done in internal, invisible space.

igiannakas, 2025-04-30:
Indeed that could work.

Not sure if there is a mechanism to detect if the void is “internal” or external though if that makes sense

vovodroid, 2025-05-01:
Slicer in general for sure knows))), probably not at this code point. But we can assume where is internal could be basing on contour/hole loop role and extrusion direction. So at least it's possible to add option "Wipe on single wall".
Or just add appropriate flag to loop where it was generated.

I tried to print such thing 

[image]

from TPU, and got blob at the each seam on unretraction.