OrcaSlicer/OrcaSlicer #9507: Add Anycubic Kobra 3 Max

robertbaker, 2025-04-28:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

missing printer

### Which printers will be beneficial to this feature?

Klipper

### Describe the solution you'd like

add profiles, can base them from their fork

SrGatoo, 2025-05-25:
Would be awesome to be able to have at least the profiles for it on orca!

davidmerel, 2025-07-03:
Same here, really looking forward to the Kobra 3 Max, so strange you have the regular Kobra 3 but no Max

tristanlins, 2025-07-20:
👍 
Same problem. Anycubic provide a linux repository, but the english domain host does not exist and the zh repo answers with access denied.

vinjaec, 2025-09-03:
> Just to complete the support chat, Anycubic has two slicers and the newer one is a little hidden.
> 
> The old "Anycubic Slicer" is based on Prusa Slicer. The new ["Anycubic Slicer Next"](https://www.anycubic.com/slicerNextDownload) is based on Orca.
> 
> Update while writing this: A few weeks ago, the Anycubic Slicer was promoted on the Website. The Next Slicer was only through the wiki. Now the Next Slicer is listed under the main menu "Software" on the Anycubic Website.
> 
> Only the Next Slicer included the Kobra 3 Max Profiles! And it works great with the Kobra 3 Max (but its a little outdated, compared to Plain Orca Slicer).

so there is still the way of copying the profile data from the nextslicer and paste it in plain orca?

tristanlins, 2025-09-15:
I "cloned" the printer preset from anycubic slicer next into orca.
[Printer presets.zip](https://github.com/user-attachments/files/22328047/Printer.presets.zip)

But I was unable to get the network connection working.
[image]
Any suggestions?

SrGatoo, 2025-09-15:
Stock printer software cannot be used online, you need to use an usb.

Second option, which I hihgly recommend is installing Rinkhals, as it gives you acces to klipper, so you can properly control the printer using any slicer.

real-limitless, 2025-10-26:
Any updates on this? 

Why hasn't anyone added this to the printer's list?
Granted, Klipper isn't available on the stock printer. but its available with Rinkals project. 

Can we at least have it as an option on the printer's selection for G-Code exports?

drc85, 2025-11-08:
i also tried to import them no luck.. 
probably will be there in of the next updates! i hope :D

tristanlins, 2025-11-09:
I would like to share my custom Anycubic Kobra 3 Max profile. I've found that the exported configuration contains absolute paths pointing to my local drive, specifically for the print bed STL model. How can I export a profile with relative or no local file system references?

robertbaker, 2025-12-05:
I will have a PR for this sometime this weekend now that Anycubic updated their firmware to officially support OrcaSlicer.

Allunga, 2025-12-15:
would also appreciate if the Kobra 3 Max profile would be added to the Orca Slicer..