OrcaSlicer/OrcaSlicer #9508: "Rectilinear interlaced" support interface pattern does not generate correctly

MrTwisT007, 2025-04-29:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

Windows

### OS Version

Windows 10

### Additional system information

Selecting "Rectilinear Interlaced" interface pattern instead generates a regular "Rectilinear" pattern. This does not seem to depend on any project of object specifically and always generates incorrectly.

### Printer

Bambu X1C

### How to reproduce

1. Enable Support
2. Set Interface pattern to "Rectilinear Interlaced"
3. Slice Plate

### Actual results

The interface pattern generated is a regular Rectilinear, with no interlacing in between.

[image]

### Expected results

A interlaced pattern to be generated. Current OrcaSlicer version does not in any circumstance generate the correct pattern, regardless of any other settings. For reference, this is an expected result exhibited by BambuStudio

[image]

### Project file & Debug log uploads

[interface_issue.zip](https://github.com/user-attachments/files/19950745/interface_issue.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

MxBrnr, 2025-04-29:
Can confirm; no interlacing in OrcaSlicer v2.3.1-dev Build e2f59f0

[image]

closed 2026-02-13 by SoftFever

MrTwisT007, 2026-02-13:
Sweet, thanks for fixing this!