OrcaSlicer/OrcaSlicer #9512: Printing Multiple Round Items with multiple Rectangular items causes a layer shift to happen.

l-Kage-l, 2025-04-29:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

Windows

### OS Version

Windows 11 IoT

### Additional system information

Doesnt Matter

### Printer

BambuLab X1C

### How to reproduce

I have a few custom made objects that are completely round, and a few custom made objects that are completely square.
1. Add them to the slicer
2. Clone them to have 8 of each object (this is not necessary? i havent tested if it reproduces with less objects)
3. Slice 
4. Nothing wrong shows up in the Preview
5. Print

### Actual results

At about layer 2/3, a layer shifts happen, moving the objects by 0.3mm to the right and 0.2mm upwards

### Expected results

No layer shift to happen.

I tested it right now, if i remove the square objects, leaving only the round ones, it prints perfectly. And so do the square ones, if printed independently

https://imgur.com/a/JIRLRbv

Github fails to upload the pictures, heres a link to them

### Project file & Debug log uploads

Files are proprietary, i cannot upload them

### Checklist of files to include

- [ ] Log file
- [ ] Project file

tlhintoq, 2025-05-06:
Layer shifts aren't the slicer - that's something mechanical in the printer.  Nozzle catching on a vertical zHop pimple or something like that.  Here' over 600 squares with round holes in the middle of each - it doesn't layer shift.

Layer shift can really only happen through software if you have the acceleration SO HIGH that the printer can't mechanically keep up and it causes a belt/motor slip

[image]

[image]

l-Kage-l, 2025-05-07:
I'm aware, I know how layer shifts work. 

The printer is bone stock, the slicer settings are bone stock. No high accelerations, nothing of such.

The printer isn't catching on a zHop pimple, it's shifting in the same exact direction in the same exact layer height every time I tried to print it, which suggests GCode errors, which suggest Slicer errors. 

That is why I created this issue.

tlhintoq, 2025-05-11:
I'd sooner guess a problem with the model.
Do you want to attach the model and have someone look at it?
And the project too?

l-Kage-l, 2025-05-11:
Hmm, why would it be a problem with the model, since the model prints individually fine, but combined on the same plate, they don't?

And sorry, I can send screenshots at best, the models are proprietary and a core part of my business. I cannot share them.

tlhintoq, 2025-05-11:
Ok.  Simple test...
Open the model in Windows 3d builder, then hit `IMPORT`. Does it show the model needs to be repaired?

[image]

l-Kage-l, 2025-05-19:
Sorry for the late reply.

Here's a screenshot from Windows 3D Builder. No Objects need repairing. They were self-designed in Fusion360.

[image]

l-Kage-l, 2025-06-22:
Coming back to this, I think I've found the issue. For whatever reason, on some multi-color prints and objects with the shapes above, you basically have to restart the printer, otherwise it will layer-shift in the exact same spot and in the exact same direction, for whatever reason. Luckily, shutting the printer down, re-slicing and sending the file fixes this, but  the bug reappears if anything else is printed and then sending a print like this

closed 2025-06-22 by l-Kage-l

l-Kage-l, 2025-07-04:
After more testing, it seems like the issue happens with the MicroLidar. Whenever a print would start, complete the first layer, scan it for issues with the MicroLidar on the X1C, then if it reports issues, after pressing Resume, it losses steps. If the microLidar doesnt see any issues, there is no step loss. Auto-recovery from step-loss helps with that, but doesnt mitigate the issue completely.