OrcaSlicer/OrcaSlicer #9520: Unable to Start OrcaSlicer appimage in KUbuntu 25.04

TheRealBeef, 2025-05-01:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

v2.3.0

### Operating System (OS)

Linux

### OS Version

KUbuntu 25.04

### Additional system information

Ryzen 5 7600x
64GB Ram
3080Ti

### Printer

Doesn't matter

### How to reproduce

Open Orcaslicer AppImage

### Actual results

It crashes (Log below)

### Expected results

It shouldn't crash

### Project file & Debug log uploads

[2025-05-01 21:39:14.156045] [0x00007fe4ea0074c0] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1
add font of NanumGothic-Regular returns 1
add font of NanumGothic-Bold returns 1

(orca-slicer:8876): Gtk-CRITICAL **: 21:39:15.486: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:8876): Gtk-CRITICAL **: 21:39:15.486: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:8876): Gtk-CRITICAL **: 21:39:15.486: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:8876): Gtk-CRITICAL **: 21:39:15.487: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:8876): Gtk-CRITICAL **: 21:39:15.487: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:8876): Gtk-CRITICAL **: 21:39:15.487: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:8876): Gtk-CRITICAL **: 21:39:15.489: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:8876): Gtk-CRITICAL **: 21:39:15.489: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:8876): Gtk-CRITICAL **: 21:39:15.489: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:8876): Gtk-CRITICAL **: 21:39:15.490: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:8876): Gtk-CRITICAL **: 21:39:15.490: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:8876): Gtk-CRITICAL **: 21:39:15.490: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

[... 235 lines cut ...]

(orca-slicer:8876): Gtk-CRITICAL **: 21:39:16.178: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:8876): Gtk-CRITICAL **: 21:39:16.179: gtk_window_resize: assertion 'height > 0' failed

(orca-slicer:8876): Gtk-CRITICAL **: 21:39:16.183: gtk_window_resize: assertion 'height > 0' failed

(orca-slicer:8876): Gtk-CRITICAL **: 21:39:16.184: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:8876): Gtk-CRITICAL **: 21:39:16.184: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:8876): Gtk-CRITICAL **: 21:39:16.184: gtk_window_resize: assertion 'height > 0' failed

(orca-slicer:8876): Gtk-CRITICAL **: 21:39:16.184: gtk_window_resize: assertion 'height > 0' failed

(orca-slicer:8876): Gtk-CRITICAL **: 21:39:16.855: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:8876): Gtk-CRITICAL **: 21:39:16.855: gtk_widget_set_size_request: assertion 'width >= -1' failed

Segmentation fault (core dumped)

### Checklist of files to include

- [x] Log file
- [ ] Project file

phpbbireland, 2025-05-02:
Similar report on my MX-Linux system.

AGTGreg, 2025-05-03:
It needs `libwebkit2gtk-4.0-dev` which is not available in your system. 

I managed to solve the issue by installing it. This package is not available in the latest repos so I added an older repo and installed it from there: 

Edit your sources file and add the repository:
`sudo nano /etc/apt/sources.list`

Add this at the end:
`deb http://gb.archive.ubuntu.com/ubuntu jammy main`

Hit `Ctr+x` to save and close. Then update and install the missing package:
`sudo apt update`
`sudo apt install libwebkit2gtk-4.0-dev`

PS: I'm running LinuxMint

phpbbireland, 2025-05-04:
libwebkit2gtk-4.0-dev was available for mx, it installed fine but no go, still same issue, will try a restart...

closed 2025-08-11 by github-actions[bot]