OrcaSlicer/OrcaSlicer #9536: Support inheritance from a user profile

johyphenel, 2025-05-04:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

https://github.com/SoftFever/OrcaSlicer/issues/7462 got closed due to inactivity, but it would still be really nice to have support for inheritance between user created profiles — so filing a new FR. 

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Copying some context from the original bug below, we'd like support for user profiles to inherit from other user created profiles, instead of being forced to inherit from preset profiles. Currently there is an error if you do this: 

`[error] 2024-11-13 07:18:55.678076[Thread 0x0000460c]:can not find parent for config ...\user\default\filament\other_profile_name.json!`

Copying from original request (https://github.com/SoftFever/OrcaSlicer/issues/7462):
"""

This is probably related to this line

[OrcaSlicer/src/libslic3r/Preset.cpp](https://github.com/SoftFever/OrcaSlicer/blob/b7bba13789a971bd5e594896e7c8d9f616bd7c4a/src/libslic3r/Preset.cpp#L1143-L1147)

Lines 1143 to 1147 in [b7bba13](https://github.com/SoftFever/OrcaSlicer/commit/b7bba13789a971bd5e594896e7c8d9f616bd7c4a)

```
 if (inherits_config) { 
     ConfigOptionString * option_str = dynamic_cast<ConfigOptionString *> (inherits_config); 
     std::string inherits_value = option_str->value; 
     inherit_preset = this->find_preset(inherits_value, false, true); 
 } else { 
```

from my limited understanding of the code base OrcaSlicer loads every preset file and checks if the inherited preset is known before knowing all presets in the folder and thus only being able to reference previously known profiles(builtin profiles).

While I know with inheritance loop issues could become an issue it would be great if the inheritet preset loading could become recursive and make it possible to use intermediate profiles instead of copying always the same settings to new files. 

"""

donmahallem, 2025-05-04:
@MxBrnr Why would it bring more problems? In PrusaSlicer it works just fine afaik. It is especially useful if you have a Single Profile for "Vendor Pla - Generic" and than just create child-profiles that just modify the color attribute for example. If something like temp/pa or whatever changes you only change it in the Generic Profile and don't have to care about all descendants.

MxBrnr, 2025-05-04:
The issue with using any kind of inheritance is if for some reason the parent profile gets corrupted or lost, it makes all the rest unusable. Additionally, the issue with continuing to use system profiles as the parent is that if the system profile gets automatically updated by the devs/contributors, then it can change those same settings in the children profiles, even if you don't want them. This has been known to cause problems for quite a few users.

I can see the benefit that you mention, but the better setup would be to make inheritance itself in _any_ form a user-selectable option. This way those who want it can set it as desired, and those who don't want it can leave it disabled.

quantumfrost, 2025-05-13:
This feature can be extremely useful when managing a bunch of profiles with small differences, especially for filament profiles which all have only minor tuning differences.

I think potential issues caused by corrupted profiles should be solved by automatic backups, not by restricting features.

Inheriting from a user profile instead would also solve the issues caused by system profiles being updated. If there is an update, the user can use the profile comparison tool and merge any differences they like

tophattwaffle, 2025-06-13:
I have the exact same use case as Johyphenel where I want to have speed differences between profiles but have everything else inherit from a "Base profile".

Really counter intuitive that an inheritance system exists but does not allow for user inheritance.

SoongJr, 2025-11-25:
I spent a week getting filament profiles set up for less redundancy by using inheritance. They _can_ be imported, so I didn't notice at first! Only on the next App startup does the inheritance get processed and the filaments simply disappear...

And now I find out this isn't supported 😣
How is everyone else managing their filaments?! What's the official suggestion? I feel a bit frustrated with what I've tried until now.

If I want to change the nozzle size on my printer, I have to keep copies of the machine profile for each size, and copies of each filament profile (with the dialed-in settings for each nozzle size).

jbeda, 2025-12-12:
I too would love to see this supported.  It would be great if there were UI too to show the inheritance tree so that folks can wrap their head around what is going on.

In the meantime, there is a workaround that I got working: In your user presets folder, put any "base" profiles in a subdir names `base`.  The relevant line of code: https://github.com/OrcaSlicer/OrcaSlicer/blob/54c876222e0fb40f6209e46ce4519bfa22b98af4/src/libslic3r/Preset.cpp#L1207

It *looks* like you can do multiple levels buy having a `base` subdir under `base` but I haven't tried that.

I'm just getting started with this stuff so I may have it wrong.

jbeda, 2026-01-04:
I'm on MacOS and have things set up at ~/Library/Application Support/OrcaSlicer/user/default/filament.

I'm still experimenting but this is what the directory tree looks like:
```
.
├── base
│   ├── PolyLite ABS @Voron 2.4 - 0.4mm.info
│   └── PolyLite ABS @Voron 2.4 - 0.4mm.json
├── PolyLite ABS @Voron 2.4 - 0.4mm - Black.info
└── PolyLite ABS @Voron 2.4 - 0.4mm - Black.json

2 directories, 4 files
```

This is what the "black" profile looks like:
```json
{
    "default_filament_colour": [
        "#000000"
    ],
    "filament_settings_id": [
        "PolyLite ABS @Voron 2.4 - 0.4mm - Black"
    ],
    "filament_shrink": [
        "99.3%"
    ],
    "from": "User",
    "inherits": "PolyLite ABS @Voron 2.4 - 0.4mm",
    "is_custom_defined": "0",
    "name": "PolyLite ABS @Voron 2.4 - 0.4mm - Black",
    "version": "2.3.1.10"
}
```

Hope that helps!